FreeCAD/FreeCAD #19400: The Addon Manager does not switching between branches properly: Fails to switch, and show the latest updates in that branch.

raindropsfromsky, 2025-02-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The Addon Manager has several problems when we switch branch.

The expected workflow is as follows:

At the beginning, show the installed version.
Let the user select another branch.
When the user presses OK, switch over to that version, and show it as currently installed branch.
From this moment, track the version status for this newly selected branch.

If we manually check the latest update, it should be able to detect the latest version in the selected branch.

But in reality, the Addon Manager fails to do any of that.

All these bugs are captured in the attached screencapture. 

[image]

--------------

Further, this status should be persistent: If we close FC and launch it again, the Addon Manager should show the last-selected branch.
Indeed, after switching to v1.7 of Ribbon WB, and restarting, the actual UI changes to v1.7.

The **Help > About** menu for Ribbon UI also confirms this.
[image]

But the Addon Manager still shows the old version (1.6.2.6)!
[image]

In other words, the Addon Manager fails to show the currect version of the addon.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/123.928
Installed mods: 
  * Beltrami 1.1.0
  * CommandPanel
  * CurvedShapes 1.0.13
  * Curves 0.6.54
  * ExplodedAssembly
  * fasteners 0.5.37
  * frame 0.1.1
  * FrameForge 0.1.2
  * FreeCAD-Ribbon 1.7.0
  * freecad.gears 1.3.0
  * FreeGrid 2.2.0
  * OpenTheme 2024.9.1
  * parts_library
  * Quetzal 1.1.0
  * SearchBar 1.3.3
  * sheetmetal 0.7.5
  * Silk 0.1.5
  * symbols_library
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-04:
did it work in the stable 1.0?
@chennes FYI

raindropsfromsky, 2025-02-04:
Frankly, while 0.2 and 1.0 were there, it just so happened that I never needed to change branches.
I have that need only recently, to test the experimental versions of Ribbon WB.

So I cannot tell for sure.
But AFAICT it was always like this.

BTW I have found [many more issues](https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aissue%20state%3Aopen%20author%3Araindropsfromsky) with the Addon Manager.

raindropsfromsky, 2025-02-04:
BTW in the screencapture, I have marked that "the Addon Manager shows that there are two updates available for v1.6.2.6."
What I mean is that when the stable branch of Ribbon WB is selected, there are two updates pending.
So if I change to the "experimental" branch, the count of pending updates should increase to three.

That does not happen.

Of course this depends on the interface logic.

For example, another option is this:
Instead of increasing the update count, the moment I change the branch, the Addon Manager could force a restart of FC.
(even before the Addon Manager itself is closed.)

But the Addon Manager simply does not acknoweledge the change of branch in any form!

chennes, 2025-02-04:
Fundamentally, I think that the underlying bug is that the addon manager is not re-reading the `package.xml` file when the `branch_changed` signal is emitted by the dialog.

chennes, 2025-02-04:
Actually, could you display the report view and re-try (with all logging enabled in user preferences, if you haven't already)? It seems like probably something is failing in the `git` call itself.

raindropsfromsky, 2025-02-05:
> could you display the report view and re-try (with all logging enabled in user preferences, if you haven't already)?
Please help me set up my FC correctly.
My settings are as follows:

[image]

Does it need any change?

> Actually, could you display the report view and re-try (with all logging enabled in user preferences, if you haven't already)? It seems like probably something is failing in the `git` call itself.

Well, I am not a coder, but when the WB is already installed and working, is Git still involved?
Even the local help is showing v1.7 but the Addon Manager shows the old version.

chennes, 2025-02-05:
> Well, I am not a coder, but when the WB is already installed and working, is Git still involved?

Yes: changing branches is a git operation. Could you please check the "Record log messages" option above, then show the report view and re-run your test, then paste the output in here?

raindropsfromsky, 2025-02-06:
I conducted three small iterations, and captured the screenshots and log.

[Addon Manager fails to show the current branch - log.odt](https://github.com/user-attachments/files/18683353/Addon.Manager.fails.to.show.the.current.branch.-.log.odt)

raindropsfromsky, 2025-02-06:
After this experiment, I used Addon Manager, and switched from 1.7 to main  branch.

The main panel of the Addon Manager shows v1.6.2.6 as "up-to-date", 
But the message shows that the current branch is **Pre-release -v1.7.0**!
That means the Addon Manager actually has the correct status, but it is unable to show it on the main screen.
[image]

chennes, 2025-02-06:
Yeah, it's probably not re-caching the package.xml file when the branch changes. The change branch feature isn't well-tested at all, thanks for digging into it and filing this report.

raindropsfromsky, 2025-02-06:
My animated image shows multiple pronblems.
Could you trace those issues from my log?
Do you need any other experiments to be carried out?
Please do let me know!

chennes, 2025-02-06:
Basically all of those bugs are the same thing: the Addon Manager is using the metadata file from the main branch, regardless of what branch you have checked out. I think you misunderstood the "updates available" button at the bottom, which applies to *all* addons, not just the one you've filtered that list down to. The two updates suggested there do not apply to the FreeCAD Ribbon addon, which is up-to-date. Therefore it will not be impacted by changing the branch at all.

chennes, 2025-02-06:
I think https://github.com/FreeCAD/FreeCAD/pull/19444 will improve the situation somewhat, at least as far as catching `git` errors is concerned. I'm leaving this bug open because I think the metadata issue will persist even after that PR is merged, but I'm still poking at how to write a regression test that captures the problem.

raindropsfromsky, 2025-02-07:
> I think you misunderstood the "updates available" button at the bottom, which applies to _all_ addons, not just the one you've filtered that list down to. The two updates suggested there do not [...]

No, I am aware of that! I even clarified this remark later as follows: 

I have chosen to check for updates automatically when the Addon Manager is launched.  At the time of my experiment, the addon Manager had already detected that two updates were available.

So, when I changed the branch, FC could have followed one of the two workflows:
1. Show that three updates pending (increase the count by one), or 
2. Deal with the branch change immediately, and let the previous two changes remain pending. 

This is an algorithmic choice. As a beta tester, I anticipated any of these options. 
But I found that neither of this was happening.

That is what I have reported.

--------------

BTW, the behavior has changed later, but not consistently: 

Just after I change the branch, FC decides that the tip of the new branch will have to be installed, and gives me an opportunity to restart. (In other words, it follows option#2 listed above.) 

But this does not happen every time.

So there is some state-dependency because of which the workflow changes. 
we need someone to analyze the code.

chennes, 2025-02-07:
When the Addon Manager runs its update check, it does a `git fetch` on each of the repos you have cloned. So at the end of that process, your local copy has all the data it needs to either switch branches, or to update your currently-checked-out branch to the latest version. From the Addon Manager's perspective, only the latter is an `update`, and the count of "Updates available" is really the check for how many repos contain the word "behind" in their `git status` output.

In your recordings, it looks to me like the "FreeCAD Ribbon" addon is up-to-date from the very beginning: that is, the current local checkout is the latest revision available on the branch that is checked out. Because the update checking process did a `git fetch`, changing to a different branch will always result in that *new* checkout being fully up-to-date (even if the branch you were on before was not). If you then change back, git will bring you to the most up-to-date revision. So as a general rule, changing branches will *never* increase the number of available updates, and in some circumstances changing from one branch to another and then back again *might* decrease the number.

The Addon Manager doesn't "remember" what branch you were on before, so basically it's going to tell you a restart is needed if you have changed branches, even if you changed and then changed back, and actually nothing "under the hood" is different. It's not that it needs to "install" anything, per se, it's that during FreeCAD's startup process it ran a file from the Addon that initializes it, and that file might have changed. On disk, the files are the files, and any git operation that needed to happen has happened. Basically the Addon is now in an unknown state, and to be safe the AM just tells you to restart.

raindropsfromsky, 2025-02-07:
> the Addon Manager is using the metadata file from the main branch, regardless of what branch you have checked out.

Well, this explains only a part of the problem I reported: 
[image]

1. It explains why the main screen shows v.1.6.2.6 (because that's the main branch).
2. It does NOT explain why the dialog tells me that the current branch is the Development branch (v1.7.0)?
    It should have stated-

     ```
     From: main 
     To: main
     ```
It looks like the main display and dialog use different sources of data!
If so, the code should be corrected.