FreeCAD/FreeCAD #19401: Position Part_DatumLine is not taken into account while reference for PolarPattern

tarman3, 2025-02-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Looks like position of Part_DatumLine is not taken into account while reference for PolarPattern in PartDesign

[image]

### Full version info

```shell
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/118.293
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.36; freecad.gears 1.3.0; fasteners 0.5.37; sheetmetal 0.7.5; btl 0.9.9; Silk 0.1.5; Curves 0.6.55; Defeaturing 1.2.3
```

### Subproject(s) affected?

None

### Anything else?

[Part_DatumLine_bug.zip](https://github.com/user-attachments/files/18656612/Part_DatumLine_bug.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-04:
Can confirm.
@PaddleStroke the placement of the DatumLine is not used, also if it is properly attached.

maxwxyz, 2025-02-04:
maybe related: https://github.com/FreeCAD/FreeCAD/issues/19060

tarman3, 2025-02-04:
Probably, Yes

closed 2025-03-07 by chennes