FreeCAD/FreeCAD #19402: Safe Mode did not return to the normal mode completelly.

leoheck, 2025-02-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Yesterday I was asked to test something on Safe Mode, and this mode is great to test things. 
After that, I started to see these files.

```bash
  280 -rw-rw-r-- 1 lheck lheck   282669 Feb  2 15:55 project.FCStd.82309420-bdff-452d-b283-81b4b64b83eb
   80 -rw-rw-r-- 1 lheck lheck    78372 Feb  3 14:46 project.FCStd.898502a3-47e3-4e24-8f66-640e49221dce
   80 -rw-rw-r-- 1 lheck lheck    78501 Feb  3 14:46 project.FCStd.f19bc17b-edb7-4c62-b073-d29050922277
   80 -rw-rw-r-- 1 lheck lheck    78504 Feb  3 14:47 project.FCStd.93f688c9-dc0a-462a-86f1-ee27745c2bdb
   80 -rw-rw-r-- 1 lheck lheck    78525 Feb  3 14:47 project.FCStd.96f27679-bb78-4270-8556-66a649952d07
   80 -rw-rw-r-- 1 lheck lheck    78526 Feb  3 14:47 project.FCStd.843d09bb-2e1d-4002-8983-3cd62947389d
   80 -rw-rw-r-- 1 lheck lheck    78546 Feb  3 14:48 project.FCStd.dde5aaf3-5a1a-460b-a97d-2575d0736d51
   80 -rw-rw-r-- 1 lheck lheck    78574 Feb  3 14:48 project.FCStd.6d282dd5-67cb-4aca-b0ea-23f01eae980d
   80 -rw-rw-r-- 1 lheck lheck    78594 Feb  3 14:49 project.FCStd.7a4ac95a-94e6-44f3-bcd5-0c864117d4c1
   80 -rw-rw-r-- 1 lheck lheck    78601 Feb  3 14:49 project.FCStd.3f37ff1f-a02c-489c-8367-5e9737f6d9d9
   80 -rw-rw-r-- 1 lheck lheck    78609 Feb  3 14:49 project.FCStd.b53c268d-9e5f-42ee-b128-9d42ed958546
```

This is a huge pile of trash.
I never used these files. They should not be generated at least in the project's folder.
I disable this feature immediately after installing Freecad.

However, the Safe Mode set it back and did not restore after leaving the Safe Mode.
Is there any button to leave the Safe Mode safely returning the user settings? 
What I did was to close Freecad and relaunch it manually.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.7.5
  * Curves 0.6.54
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-04:
@hyarion is this safe mode related? What info is needed?

hyarion, 2025-02-04:
@leoheck You are only describing what you're seeing, I need:
1. steps to reproduce the issue (and test the steps before answering please)
2. path to where you found the files.

To exit safe mode you have to restart the application

leoheck, 2025-02-08:
Alright, this is not due to the safe mode I guess now.

My settings are as they used to be.

[image]

But I am still seeing those files getting generated.

```bash
20396 -rw-rw-r-- 1 lheck lheck 20881791 Feb  8 11:04 Body.FCStd
  180 -rw-rw-r-- 1 lheck lheck   181555 Feb  7 16:20 Body.FCStd.0e5d9ca4-7ca9-46c7-907b-16c9497beaac
  192 -rw-rw-r-- 1 lheck lheck   193543 Feb  7 16:20 Body.FCStd.d8ebbc45-b19a-4137-b462-ab0096a1483f
```

Maybe these are generated now due to segfaults.
I am experiencing a lot of segfaults working with 1.1-dev.

Those files both have only the `Document.xml` inside.

hyarion, 2025-02-08:
> Alright, this is not due to the safe mode I guess now.

Sounds like this issue can be closed then.

@maxwxyz ping

closed 2025-02-09 by maxwxyz

leoheck, 2025-02-09:
Actually no, these files should not be generated since they are useless.

hyarion, 2025-02-09:
These files are created due to your settings.
The location depends on where the file loaded is located.

Safe mode doesn't mean that it won't touch the file system, just that it temporarily works as a fresh install.

This post explains what you are seeing https://forum.freecad.org/viewtopic.php?style=3&t=12345

leoheck, 2025-02-09:
This is a thread from 2015, that ends in 2020. The current year is 2025. Freecad crashes a lot, multiple times a day. And last week is the first time seeing these files.