FreeCAD/FreeCAD #19408: Crash

zdavatz, 2025-02-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

1. running this script:
```
import FreeCAD
import Part
import numpy as np

# Main parameters
span = 1200.0       # mm (110 cm)
root_chord = 230.0  # mm (23 cm)
tip_chord = 90.7    # mm (17.07 cm)
naca_thickness = 0.12  # Thickness ratio for NACA 0012
num_points = 100

# Gull-wing parameters
bend_start = 350    # Start transition at 350mm
bend_end = 750      # End transition at 750mm
root_dihedral = 0   # Degrees (root section)
mid_dihedral = 10   # Degrees (maximum bend)
tip_dihedral = 5    # Degrees (tip section)

def create_airfoil_section(chord, thickness, y_pos, z_offset):
    """Generate NACA 0012 airfoil with vertical offset"""
    points = []
    
    # Upper surface points
    for x_norm in np.linspace(0, 1, num_points):
        x = x_norm * chord
        yt = (thickness / 0.2) * (0.2969 * np.sqrt(x_norm) - 0.1260 * x_norm -
              0.3516 * x_norm**2 + 0.2843 * x_norm**3 - 0.1015 * x_norm**4)
        points.append(FreeCAD.Vector(x, y_pos, yt * chord + z_offset))
    
    # Lower surface points
    for x_norm in np.linspace(1, 0, num_points)[1:]:
        x = x_norm * chord
        yt = (thickness / 0.2) * (0.2969 * np.sqrt(x_norm) - 0.1260 * x_norm -
              0.3516 * x_norm**2 + 0.2843 * x_norm**3 - 0.1015 * x_norm**4)
        points.append(FreeCAD.Vector(x, y_pos, -yt * chord + z_offset))
    
    points.append(points[0])
    return Part.makePolygon(points)

# Create document
doc = FreeCAD.newDocument("Gull Wing")
sections = []
z_offsets = []

# Calculate dihedral transition
y_positions = np.linspace(0, span, 15)
prev_z = 0
prev_y = 0

for y in y_positions:
    if y <= bend_start:
        angle = root_dihedral
    elif y < bend_end:
        t = (y - bend_start) / (bend_end - bend_start)
        angle = root_dihedral + t * (mid_dihedral - root_dihedral)
    else:
        t = (y - bend_end) / (span - bend_end)
        angle = mid_dihedral + t * (tip_dihedral - mid_dihedral)
    
    delta_y = y - prev_y
    z_offset = prev_z + delta_y * np.tan(np.radians(angle))
    chord = root_chord + (tip_chord - root_chord) * (y / span)
[... 130 lines cut ...]
#26  0x7fe4d361e712 in Gui::MacroManager::run(Gui::MacroManager::MacroType, char const*) from /app/freecad/lib/libFreeCADGui.so+0x1d2
#27  0x7fe4d38e5cd7 in Gui::PythonEditorView::executeScript() from /app/freecad/lib/libFreeCADGui.so+0xd7
#28  0x7fe4d38e5e78 in Gui::PythonEditorView::onMsg(char const*, char const**) from /app/freecad/lib/libFreeCADGui.so+0x98
#29  0x7fe4d352d94f in Gui::Application::sendMsgToActiveView(char const*, char const**) from /app/freecad/lib/libFreeCADGui.so+0x3f
#30  0x7fe4d3569f04 in Gui::Application::sSendActiveView(_object*, _object*) from /app/freecad/lib/libFreeCADGui.so+0x84
#31  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1c3116) [0x7fe4d20b4116]
#32  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x97) [0x7fe4d20b3f17]
#33  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x2db6) [0x7fe4d1ff7196]
#34  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2d2176) [0x7fe4d21c3176]
#35  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyEval_EvalCode+0xaa) [0x7fe4d21c322a]
#36  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2d2b24) [0x7fe4d21c3b24]
#37  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2d2c1b) [0x7fe4d21c3c1b]
```

### Subproject(s) affected?

None

### Anything else?

Let me know if you want me to test something.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-06:
Does it work on the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds ?
Please add full about info.

zdavatz, 2025-02-07:
[image]

maxwxyz, 2025-02-07:
@zdavatz workaround: https://github.com/FreeCAD/FreeCAD-Bundle/issues/360#issuecomment-2561365902