FreeCAD/FreeCAD #19410: PartDesign: The Pad does not work properly when growing inside a hollow box.

leoheck, 2025-02-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The Pad does not work properly when growing on a hollow but closed box.

[Pad_Issues.FCStd.zip](https://github.com/user-attachments/files/18661865/Pad_Issues.FCStd.zip)

See the last feature `Pad005` turns the hollow compartment filled.

The file is currently this way and filled... 
[image]

When inverting `Pad005` direction (as intended by the user) it becomes this (still filled)
[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.7.5
  * Curves 0.6.54
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-02-05:
AdditiveLoft does the same.

Here I have 3 Sketches to make a screw boss, inside of a hollow box.
[image]

Here is another angle.
[image]

The object added by the loft is projected correctly.

[image]

However, after applying the operation destroys the box (filling the internal space, and removing the shell)
[image]

Here is another angle after the operation.

[image]

The same result happens when moving sketches far from coincident points.

[image]

FlachyJoe, 2025-02-05:
The problem come from _Closing_Bosses_Ribs_ where the refinement break the shape. Switching `Closing_Bosses_Ribs.Refine` to False solve the issue.

FEA-eng, 2025-09-29:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```