FreeCAD/FreeCAD #19415: External Projection Geometry not staying synced when body updated by Spreadsheet

steveklett, 2025-02-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Updating parameters of a feature (e.g., Pad) with a Spreadsheet will not correctly update External Project Geometry. The behavior can be different, sometimes it will update if the Pad increases in size and will "push" the external geometry out as it grows, then "leave it behind" when the pad is reduced in size.

I have created a video demonstrating the issue: https://youtu.be/DGBH41MiL-A and a longer one: https://youtu.be/QHCPkVBVeng

### Reproduction Steps (from scratch)
See attached file: `external_geom_issue_repro1.FCStd`

1. Create new Part Design, Body and Sketch on XZ plane
2. Draw a rectangle and complete the sketch
3. Pad to form a part, set Symmetric to plane checked
4. Select the face on the ZY plane of the part and add a sketch
5. Create External Project Geometry and select the top edge of the part (an edge that will shrink/expand as the Pad length is changed)
6. Make the external geometry Construction Geometry
[image]

7. Close the Sketch
8. Change to Spreadsheet WB and create a new Spreadsheet
9. Set A1 to =1in (or whatever unit) and set an Alias to `pad_length`
10. Switch back to Part Design
11. Select the Pad in the Tree and set Length to expression `Spreadsheet.pad_length`
12. Tile your windows so you can see the spreadsheet and 3D view side by side (Windows > Tile)
13. 2x click on the Sketch in the tree to edit it
14. Set spreadsheet cell A1 to 2in and observe the sketch - the construction geometry doesn't update
[image]
15. Test

### Reproduction Steps (More complex, but demonstrates how bad this bug can be)
This is the file I was working on when I discovered this issue. It is more complex and I didn't want to make it the only way to reproduce this issue (see above). I'm including it here because it demonstrates how this bug can thoroughly break a drawing.
See attached file: `external_geom_issue_repro2.FCStd`

1. Open the file `external_geom_issue_repro2.FCStd`
2. Open the Spreadsheet
3. Tile the windows so you can see both side by side
[image]

4. **Note** the white rectangles in the 3D view, these are used to pocket the grooves in the green panel. The faces created by the pocket are then extracted as external geometry and cloned into other Bodies to eventually create the Lofts for the 3 horizontal shelves. Point is, there is a lot of downstream dependencies on those 3 white triangles.
5. Edit Spreadsheet cell `B4` to `2in`, observe that everything seems to work fine (panel thickness increases, etc.)
[image]

6. Now edit cell `B4` to `.82in` and get ready for the errors!
```
12:41:38  <PropertyLinks> PropertyLinks.cpp(491): bookcase___master_sketch___WIP_projection_rnd___manually_cut_left_side#AdditiveLoft.Sections missing element reference bookcase___master_sketch___WIP_projection_rnd___manually_cut_left_side#Binder001 ;#38:5;:M;CUT;:H-6e1:7,F;:H6e1,F.Face10
12:41:38  <PropertyLinks> PropertyLinks.cpp(491): bookcase___master_sketch___WIP_projection_rnd___manually_cut_left_side#AdditiveLoft001.Sections missing element reference bookcase___master_sketch___WIP_projection_rnd___manually_cut_left_side#Binder002 ;#38:f;:M;CUT;:H-6e1:7,F;:H6e1,F.Face14
12:41:38  <PropertyLinks> PropertyLinks.cpp(491): bookcase___master_sketch___WIP_projection_rnd___manually_cut_left_side#AdditiveLoft002.Sections missing element reference bookcase___master_sketch___WIP_projection_rnd___manually_cut_left_side#Binder003 ;#38:12;:M;CUT;:H-6e1:7,F;:H6e1,F.Face6
12:41:38  top shelf: Failed to get shape of Section Binder001.?Face10
12:41:38  middle shelf: Failed to get shape of Section Binder002.?Face14
12:41:38  bottom shelf001: Failed to get shape of Section Binder003.?Face6
```

[image]

**

[external_geom_issue_repro1.FCStd.tar.gz](https://github.com/user-attachments/files/18663278/external_geom_issue_repro1.FCStd.tar.gz)

[external_geom_issue_repro2.FCStd.tar.gz](https://github.com/user-attachments/files/18663279/external_geom_issue_repro2.FCStd.tar.gz)

**

1. The external Projection Geometry created from the edge of the green panel did not sync/track with the Pad changes triggered by the Spreadsheet update.
2. This resulted in the blue Construction Geometry line that the 3 rectangles used for the Pocket being constrained to the opposite side of the red Construction Geometry
[image]

3. This results in t he Pocket failing which means the faces referenced by the Binders are missing

### Full version info

```shell
OS: Linux Mint 22 (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96.0135/108.888
```

### Subproject(s) affected?

None

### Anything else?

I put a lot of detail in the main Issue body, but there is also a UG post, though at the time of this writing it's not getting much attention.
https://forum.freecad.org/viewtopic.php?p=809058#p809058

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

steveklett, 2025-02-04:
I believe I have a workaround for my particular case, though I'm not sure if there are other issues with the file I'm not yet aware of.

1. Deactivate the constraint that puts the construction line on the right of the external geometry
2. Drag it over to the side it should be on
3. Reactivate the constraint
4. It will snap into the correct location and the part will generate without errors

Video demonstrating:
https://youtu.be/mr__UoIw-CY

maxwxyz, 2025-02-05:
@PaddleStroke could you please take a look at this?

Roy-043, 2025-02-05:
It seems to me that these are two distinct issues.

The first is a recompute issue: while the sketch is in edit mode external geometry is not updated if referenced objects change.

The second is a typical flipping problem I think.

steveklett, 2025-02-05:
@Roy-043 After experimenting more I agree it seems to only be a recompute issue. I constrained a rectangle to that construction line so I could see the changes and closed the sketch. Updating the spreadsheet keeps the external geometry line synced. So my bad reporting this as a problem.

As for the typical flipping problem, I'm trying to understand why that isn't a bug. If the second construction line is constrained to the right of the first, how could that ever change during correct behavior and operation? As long as the driving element of the constraint behaves correctly, shouldn't the driven/constrained element always remain in the same relative position?

Roy-043, 2025-02-13:
Flipping is really annoying. It typically occurs after 'big' changes to the sketch. At its core there is irrefutable logic: certain constraint schemes have multiple solutions and the solver stops at the first solution it finds.

Two known ways to reduce the chance of flipping:
* Use angular constraints. Using them over horizontal and vertical constraints can work in some cases.
* Switch to a different solver. The [LevenbergMarquardt](https://wiki.freecad.org/Sketcher_Dialog#Advanced_solver_control) solver is less likely to flip sketches (but you are already using that solver).

I could not find a solution for this particular case before, but I'll look at your file again.

Roy-043, 2025-02-13:
OK, I think I have found a workaround. I have attached a sketch with a rectangle to a face of Pad001 in the "Right Panel" Body. This sketch is attached with an offset `-Spreadsheet.rabbet_depth`. One of its edges is referenced as external geomery in the "dados and rabbets" sketch.

Instead of a sketch I actually wanted use a datum plane, but using that for external geometry does not seem to work in V1.1.

[external_geom_issue_repro2_roy2.FCStd.zip](https://github.com/user-attachments/files/18787925/external_geom_issue_repro2_roy2.FCStd.zip)

maxwxyz, 2025-02-14:
I'll close this issue then, as recompute solves this.

closed 2025-02-14 by maxwxyz