FreeCAD/FreeCAD #19416: FreeCAD will not generate gcode

Lin1Jay1, 2025-02-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have a program for a tray that I have made previously made using FreeCAD.  This tray was my 1st project used to learn FreeCAD.  I have 30 years experience with SolidWorks so I had no problem learning to model the tray.  I have had a lot of problems with CAM and have learned it's early days for CAM.  The tray has two parts, an outer frame and an inner insert.  I wanted to revise the design subtly to make the frame material (not the frame itself) wider so it would be more prominent visually and the insert would be a bit smaller.

[image]

I saved the file with a new (Wide Frame) name and made appropriate subtle changes to the model.  If this were SolidWorks  this would be a simple creation of a new configuration in the original file.
I then began inspecting / checking / fixing any problems in the CAM operations.  I reached a profile operation that showed the tool path but will not generate gcode.  I tried creating that tool path again as a separate operation and I got exactly the same result - I can see the tool path but no gcode.

[12x18x0.94 Inlay W Frame-MachTop-2.3.25-6 prob.FCStd.zip](https://github.com/user-attachments/files/18662872/12x18x0.94.Inlay.W.Frame-MachTop-2.3.25-6.prob.FCStd.zip)
The operation 3-Profile inner-Remove excess frame 1.14D x 0.62 Face Mill To -0.5625 shows the correct tool path and the simulation runs correctly.

[image]
But when I try to generate gcode I just get error messages,

[image]
I stripped out all the operations after 2-Face top rim out to in.  Same problem.  I can see the path and simulation run correctly but no gcode

[12x18x0.94 Inlay W Frame-MachTop-2.4.25-2 OK2.FCStd.zip](https://github.com/user-attachments/files/18663051/12x18x0.94.Inlay.W.Frame-MachTop-2.4.25-2.OK2.FCStd.zip)
The 1st operation will generate gcode for the dowel holes required.
I opened the file in RC2 but had the same problem.
I noticed there seems to be a problem with file size:

[image]
In a matter of minutes, the file size ballooned from 805 KB to 3216 KB.  

The one thing I really don't understand about FreeCAD is all the models that are created.  It just seems to cause a lot of unnecessary problems.
I'm used to using spreadsheets to create configurations in SolidWorks and perhaps I can use this as a work around on this project.  Right now I'm wishing I'd done this project using Carveco Maker.  I didn't really like Maker because it was useless for design purposes and you had to know material size in advance and you were locked into that size.  But if I imported DFX files from SWX it was very easy to generate the tool paths required.  It was pretty efficient.  FreeCAD has turned out to require hours and hours more time to do something that should be relatively simple.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * btl 0.9.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-02-04:
Answered in the forum
https://forum.freecad.org/viewtopic.php?t=94531

phaseloop, 2025-03-17:
I remember this bug happening a lot but I haven't seen it using dev versions released somewhere after 1.0.0. It is possible it was fixed one way or another.

closed 2025-07-05 by sliptonic