FreeCAD/FreeCAD #19417: CAM G02 outputs invalid parameter K0.000 with refactored_grbl_post.py

PatrikRindlisbacher, 2025-02-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

CAM Profile outputs **invalid** parameter K0.000 with refactored_grbl_post.py

**G2** X122.959 Y23.227 Z21.000 I-7.410 J-10.686 **K0.000** F1000.000

**G17** G90 G21
For the YX level, the K parameter doesn't make any sense either, as it is intended for the ?Z level

Unfortunately the Eding CNC machine doesn't work like this

### Full version info

```shell
OS: Windows 11 build 22621
Architecture: x86_64
Version: 1.1.0dev.40041 (Git) Conda
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/93.1131
```

### Subproject(s) affected?

None

### Anything else?

--> Arcs may only work on the XY plane (this needs to be verified).
For refactored_linuxcnc_post.py the parameter "K" was simply deleted from the list instead of fixing the error

The parameter K should only be suppressed for plane XY  --> G17

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

LarryWoestman, 2025-02-04:
To the best of my knowledge, the current FreeCAD code only supports arcs on the XY plane.  That limitation is noted in comments in the code.

The refactored* postprocessors are still "under construction".  If you use them you will be exposed to a lot of changes.

What can I do to reproduce this issue?  You show the output of the postprocessor but not the "input" in the form of a FreeCAD file or other description of what to do to make this happen.

Please assign this issue to me.

PatrikRindlisbacher, 2025-02-05:
Thanks for your interest in the error

The error occurs with every circular movement of a round contour, for example.
- refactored_grbl_post.py

false --> G2 X122.959 Y23.227 Z21.000 I-7.410 J-10.686 K0.000 F1000.000
true  --> G2 X122.959 Y23.227 Z21.000 I-7.410 J-10.686 F1000.000

PatrikRindlisbacher, 2025-02-05:
Is this the information you were looking for?

LarryWoestman, 2025-02-06:
I will have to wait for detailed analysis until I have time to create a FreeCAD file with a drawing of an object that has a curve and a corresponding CAM job.  Not a big deal, but a bit time consuming.  Hopefully any such file will show the issue.

As a workaround, you could edit the refactored_grbl_post.py file and delete the "K" at line 108 (or so) from the "Values["Parameter_Order"] variable.  That will cause any K parameters to be ignored by the postprocessor.

PatrikRindlisbacher, 2025-02-06:
One circle, one 

body, one operation... and the error is reproducible

PatrikRindlisbacher, 2025-02-06:
[G02.zip](https://github.com/user-attachments/files/18697828/G02.zip)

LarryWoestman, 2025-02-07:
I was able to use your FreeCAD file to confirm your results.  Thank you for the file, it made things easier.

I was able to track down the code in FreeCAD that generates the arc "internal data structure".  It is generating I, J, and K parameters, so that matches.

As far as I can tell from online documentation, the GRBL code base should be able to understand a K parameter, but it is highly likely that it, like LinuxCNC, doesn't want to see a K parameter for an XY plane arc, just I and J.

So I conclude that the existing grbl and refactored_grbl code should not output a K parameter for an XY plane arc, which again matches what you reported.

I consider this a bug in the refactored code base and the original grbl_post code base.  Based on my understanding of those code bases, the existing grbl postprocessors don't "understand" G17/G18/G19 and always assume G17 (the XY plane) for any arcs.

The workaround of removing the "K" from the list of parameters to pay attention to should not have any other side effects that I could see.

I will add this issue to my list of "things that need to be fixed", and hope to get a pull request to fix this done "fairly soon".

Please assign this issue to me.

PatrikRindlisbacher, 2025-02-07:
I still have to get used to the processes and terms in Github.

You write: 
Please point this issue to my attention.

What do i have to do? @LarryWoestman

LarryWoestman, 2025-02-07:
You don't need to do anything, but one of the people who manages the issues will hopefully assign the issue to me since I don't appear to have the permissions to assign it myself.

LarryWoestman, 2025-02-17:
@maxwxyz Would you please assign this issue to me.  I don't appear to have any way to do it myself.  Thanks!