FreeCAD/FreeCAD #19425: Sketcher:  Freecad crashes when trimming edges.

leoheck, 2025-02-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Crash when using Trim Edge.

[trim_edge_issue.FCStd.zip](https://github.com/user-attachments/files/18672541/trim_edge_issue.FCStd.zip)

I have this shape that is getting complex.

[image]

Then I decided to make 1/4 and then use the transform feature later.
The quarter I want to preserve is the one with positive coordinates.

[image]

[image]

Once I start using Trim Edge to cut edges on the other quarters of the image freecad crashes.

This output happened when it crashed inside of my complex project.

```
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
<Gui> Document.cpp(1309): Physical path: /home/lheck/Dropbox/deepx/Echo54/Ground-Sensor/Case/v1.3/Body.FCStd
<Gui> Document.cpp(1311):   Document: Body: Body.FCStd
<Gui> Document.cpp(1311):   Document: Body (Body1): Body.FCStd
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7c2748645320]
#1  0x7c270a48e19e in Part::Geometry::hasExtension(Base::Type const&) const from /tmp/.mount_FreeCAHeNGek/usr/lib/./Part.so+0x1ee
#2  0x7c26e8843bec in Sketcher::GeometryFacade::initExtension() from /tmp/.mount_FreeCAHeNGek/usr/lib/Sketcher.so+0x3c
#3  0x7c26e8843ffb in Sketcher::GeometryFacade::GeometryFacade(Part::Geometry const*, bool) from /tmp/.mount_FreeCAHeNGek/usr/lib/Sketcher.so+0x5b
#4  0x7c26e8844045 in Sketcher::GeometryFacade::getFacade(Part::Geometry const*, bool) from /tmp/.mount_FreeCAHeNGek/usr/lib/Sketcher.so+0x35
#5  0x7c26e8845691 in Sketcher::GeometryFacade::copyId(Part::Geometry const*, Part::Geometry*) from /tmp/.mount_FreeCAHeNGek/usr/lib/Sketcher.so+0x41
#6  0x7c26e8728a55 in Sketcher::SketchObject::trim(int, Base::Vector3<double> const&) from /tmp/.mount_FreeCAHeNGek/usr/lib/Sketcher.so+0xd15
#7  0x7c26e87e8cf3 in Sketcher::SketchObjectPy::trim(_object*) from /tmp/.mount_FreeCAHeNGek/usr/lib/Sketcher.so+0x93
#8  0x7c26e87e9114 in Sketcher::SketchObjectPy::staticCallback_trim(_object*, _object*) from /tmp/.mount_FreeCAHeNGek/usr/lib/Sketcher.so+0x24
#9  /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7c274b006068]
#10  /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7c274afe5b5c]
#11  /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7c274aff1532]
#12  /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7c274b0abd1d]
#13  /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7c274b0ab794]
#14  /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7c274b0cabf3]
#15  /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7c274b0c669a]
#16  /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7c274b0ba9c9]
#17  0x7c274a3477cc in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/libFreeCADBase.so+0x6c
#18  0x7c274b9c4ef4 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/libFreeCADGui.so+0x94
#19  0x7c274b9c5083 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/libFreeCADGui.so+0xe3
#20  0x7c263a7f5cd2 in void Gui::cmdAppObjectArgs<int&, double&, double&>(App::DocumentObject const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&, double&, double&) from /tmp/.mount_FreeCAHeNGek/usr/lib/./SketcherGui.so+0x1a2
#21  /tmp/.mount_FreeCAHeNGek/usr/lib/./SketcherGui.so(+0x208954) [0x7c263a808954]
#22  /tmp/.mount_FreeCAHeNGek/usr/lib/./SketcherGui.so(+0x208a8d) [0x7c263a808a8d]
#23  0x7c263a974356 in SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) from /tmp/.mount_FreeCAHeNGek/usr/lib/./SketcherGui.so+0x626
#24  0x7c274bde3370 in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/libFreeCADGui.so+0x160
#25  0x7c27480d346e in SoEventCallback::handleEvent(SoHandleEventAction*) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/./libCoin.so.80+0xbe
#26  0x7c2747ed54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/./libCoin.so.80+0x68
#27  0x7c274808cb9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/./libCoin.so.80+0xea
#28  0x7c27480dc348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/./libCoin.so.80+0x58
#29  0x7c27480f713a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/./libCoin.so.80+0x2a
#30  0x7c2747ed54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/./libCoin.so.80+0x68
#31  0x7c274808cb9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/./libCoin.so.80+0xea
#32  0x7c27480dc348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/./libCoin.so.80+0x58
#33  0x7c27480f713a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/./libCoin.so.80+0x2a
#34  0x7c2747ed54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/./libCoin.so.80+0x68
#35  0x7c2747edf26c in SoHandleEventAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/./libCoin.so.80+0xac
#36  0x7c2747ed61ba in SoAction::apply(SoNode*) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/./libCoin.so.80+0x2ea
#37  0x7c27480a8ad2 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/./libCoin.so.80+0x72
#38  0x7c274bd63a9e in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAHeNGek/usr/bin/../lib/libFreeCADGui.so+0x1e
[... 46 lines cut ...]
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.7.5
  * Curves 0.6.54
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-05:
Maybe related: https://github.com/FreeCAD/FreeCAD/issues/19396
@AjinkyaDahale FYI

EstebanFPV, 2025-02-05:
It was also refusing to trim some edges for me. seemed circles and arcs were ok, straight lines fail.

AjinkyaDahale, 2025-02-05:
Could you please confirm if it is present with https://github.com/FreeCAD/FreeCAD/pull/18916?

maxwxyz, 2025-02-10:
Please test with the new weekly tomorrow: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Shkolik, 2025-02-26:
For sure related to #19770

tested on latest main - still failing, but no hard crash.

AjinkyaDahale, 2025-03-01:
Just tested. Indeed this is a problem, and glad that at least there's no crash anymore. The symmetric constraint does seem to be a question though. Could someone check with 1.0 how it behaves?

leoheck, 2025-03-04:
It is still crashing for me. 

```
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda AppImage
```

```
External_Polishing: BRep_API: command not done
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x7dc4df445330]
#1  0x7dc4772e1af7 in Sketcher::SketchObject::isClosedCurve(Part::Geometry const*) from /tmp/.mount_FreeCAdhDNkF/usr/lib/Sketcher.so+0x17
#2  0x7dc47732ee11 in Sketcher::SketchObject::trim(int, Base::Vector3<double> const&) from /tmp/.mount_FreeCAdhDNkF/usr/lib/Sketcher.so+0x7d1
#3  0x7dc4773efa73 in Sketcher::SketchObjectPy::trim(_object*) from /tmp/.mount_FreeCAdhDNkF/usr/lib/Sketcher.so+0x93
#4  0x7dc4773efe94 in Sketcher::SketchObjectPy::staticCallback_trim(_object*, _object*) from /tmp/.mount_FreeCAdhDNkF/usr/lib/Sketcher.so+0x24
#5  /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7dc4e1e06068]
#6  /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7dc4e1de5b5c]
#7  /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7dc4e1df1532]
#8  /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7dc4e1eabd1d]
#9  /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7dc4e1eab794]
#10  /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7dc4e1ecabf3]
#11  /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7dc4e1ec669a]
#12  /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7dc4e1eba9c9]
#13  0x7dc4e114ceac in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libFreeCADBase.so+0x6c
#14  0x7dc4e27d08e4 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libFreeCADGui.so+0x94
#15  0x7dc4e27d0a73 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libFreeCADGui.so+0xe3
#16  0x7dc3d17f709b in void Gui::cmdAppObjectArgs<int&, double&, double&>(App::DocumentObject const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&, double&, double&) from /tmp/.mount_FreeCAdhDNkF/usr/lib/./SketcherGui.so+0x18b
#17  /tmp/.mount_FreeCAdhDNkF/usr/lib/./SketcherGui.so(+0x1f7364) [0x7dc3d17f7364]
#18  /tmp/.mount_FreeCAdhDNkF/usr/lib/./SketcherGui.so(+0x1f749d) [0x7dc3d17f749d]
#19  0x7dc3d1964bf6 in SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) from /tmp/.mount_FreeCAdhDNkF/usr/lib/./SketcherGui.so+0x626
#20  0x7dc4e2bf3bb2 in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libFreeCADGui.so+0x162
#21  0x7dc4deed346e in SoEventCallback::handleEvent(SoHandleEventAction*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/./libCoin.so.80+0xbe
#22  0x7dc4decd54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/./libCoin.so.80+0x68
#23  0x7dc4dee8cb9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/./libCoin.so.80+0xea
#24  0x7dc4deedc348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/./libCoin.so.80+0x58
#25  0x7dc4deef713a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/./libCoin.so.80+0x2a
#26  0x7dc4decd54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/./libCoin.so.80+0x68
#27  0x7dc4dee8cb9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/./libCoin.so.80+0xea
#28  0x7dc4deedc348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/./libCoin.so.80+0x58
#29  0x7dc4deef713a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/./libCoin.so.80+0x2a
#30  0x7dc4decd54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/./libCoin.so.80+0x68
#31  0x7dc4decdf26c in SoHandleEventAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/./libCoin.so.80+0xac
#32  0x7dc4decd61ba in SoAction::apply(SoNode*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/./libCoin.so.80+0x2ea
#33  0x7dc4deea8ad2 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/./libCoin.so.80+0x72
#34  0x7dc4e2b70f8e in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libFreeCADGui.so+0x1e
#35  0x7dc4e2b7c948 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libFreeCADGui.so+0x58
#36  0x7dc4e2bd4b60 in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libFreeCADGui.so+0x40
#37  0x7dc4e2bde31f in Gui::TouchpadNavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libFreeCADGui.so+0x2df
#38  0x7dc4e2bd64c5 in Gui::NavigationStyle::processEvent(SoEvent const*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libFreeCADGui.so+0x65
#39  0x7dc4e2b9ec1b in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libFreeCADGui.so+0x6b
[... cut ...]
#60  0x7dc4dfabd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libQt5Core.so.5+0x82
#61  0x7dc4e26decad in Gui::Application::runApplication() from /tmp/.mount_FreeCAdhDNkF/usr/bin/../lib/libFreeCADGui.so+0x70d
#62  /tmp/.mount_FreeCAdhDNkF/usr/bin/freecad(+0x7906) [0x5f2831ccb906]
#63  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x7dc4df42a1ca]
#64  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7dc4df42a28b]
#65  /tmp/.mount_FreeCAdhDNkF/usr/bin/freecad(+0x7c8e) [0x5f2831ccbc8e]
```

AjinkyaDahale, 2025-03-04:
@leoheck is the crash occasional or consistent? What is the order in which you trim?

leoheck, 2025-03-05:
@AjinkyaDahale it works most of the time, and eventually it crashes. It got improved after the discussion here. Since I was busy working to finish a project, I could not analyze the sketch better. But if I am not mistaken, it always has issues when trimming circles. This is mostly what I have been trimming recently. So, I also think that in the last crash reported above I tried to trim using a blue line (I don't know how to call it, the one that is not the construction mode)

A related issue with the trim tool when doing it with circles is that it sometimes removes more than it should. But using ctrl+z  sometimes fixes the issue. I am not sure if this is part of the solution, or thinking now... it is doing twice on a click, or perhaps my (Bluetooth) mouse is clicking twice, but nothing like this happens with other tools.

AjinkyaDahale, 2025-03-08:
Previous result as from 1.0 (courtesy @casterle): 

[trim_edge_issue_Post_Trim_in_1.0.FCStd.zip](https://github.com/user-attachments/files/19145046/trim_edge_issue_Post_Trim_in_1.0.FCStd.zip)

leoheck, 2025-04-03:
Trim still crashes FreeCAD.

This always tends to happen when working with circles/arcs.

Here is the sample file [trim_crash.FCStd.zip](https://github.com/user-attachments/files/19588341/trim_crash.FCStd.zip)

When I try to trim the horizontal line, no matter where (inside or outside of the arc, my app crashed.

[image]

```
~/Applications …
➜ ./freecad_weekly_41127 
FreeCAD 1.1.0, Libs: 1.1.0devR41127 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

OSHAutoDoc workbench (v0.2.3) loaded
Initializing Assembly4 workbench (0.50.19-dev) ............................. done.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x7d6688845330]
#1  0x7d658f2dac67 in Sketcher::SketchObject::isClosedCurve(Part::Geometry const*) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/lib/Sketcher.so+0x17
#2  0x7d658f3276d3 in Sketcher::SketchObject::trim(int, Base::Vector3<double> const&) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/lib/Sketcher.so+0x753
#3  0x7d658f3f7e33 in Sketcher::SketchObjectPy::trim(_object*) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/lib/Sketcher.so+0x93
#4  0x7d658f3f80a4 in Sketcher::SketchObjectPy::staticCallback_trim(_object*, _object*) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/lib/Sketcher.so+0x24
#5  /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x1fae88) [0x7d668b1fae88]
#6  /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x2b4) [0x7d668b1d9b84]
#7  /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6e4) [0x7d668b1e5314]
#8  /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x29ee6d) [0x7d668b29ee6d]
#9  /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7d668b29e864]
#10  /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x2bce33) [0x7d668b2bce33]
#11  /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x2b8a4a) [0x7d668b2b8a4a]
#12  /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7d668b2acf29]
#13  0x7d668a550508 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/libFreeCADBase.so+0x68
#14  0x7d668bbf3154 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0x94
#15  0x7d668bbf32df in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0xdf
#16  0x7d65869f5fa6 in void Gui::cmdAppObjectArgs<int&, double&, double&>(App::DocumentObject const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&, double&, double&) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/lib/./SketcherGui.so+0x186
#17  /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/lib/./SketcherGui.so(+0x1f625d) [0x7d65869f625d]
#18  /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/lib/./SketcherGui.so(+0x1f638d) [0x7d65869f638d]
#19  0x7d6586b61846 in SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/lib/./SketcherGui.so+0x626
#20  0x7d668c00da12 in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0x162
#21  0x7d66882d346e in SoEventCallback::handleEvent(SoHandleEventAction*) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/./libCoin.so.80+0xbe
#22  0x7d66880d54e8 in SoAction::traverse(SoNode*) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x68
#23  0x7d668828cb9a in SoChildList::traverse(SoAction*, int, int) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/./libCoin.so.80+0xea
#24  0x7d66882dc348 in SoGroup::doAction(SoAction*) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x58
#25  0x7d66882f713a in SoSeparator::doAction(SoAction*) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x2a
#26  0x7d66880d54e8 in SoAction::traverse(SoNode*) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x68
#27  0x7d668828cb9a in SoChildList::traverse(SoAction*, int, int) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/./libCoin.so.80+0xea
#28  0x7d66882dc348 in SoGroup::doAction(SoAction*) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x58
#29  0x7d66882f713a in SoSeparator::doAction(SoAction*) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x2a
#30  0x7d66880d54e8 in SoAction::traverse(SoNode*) from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x68
[... cut ...]
#61  0x7d668baf9f24 in Gui::Application::runApplication() from /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0x734
#62  /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/freecad(+0x5ca0) [0x624e6be2aca0]
#63  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x7d668882a1ca]
#64  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7d668882a28b]
#65  /home/lheck/Applications/FreeCAD-41127/squashfs-root/usr/bin/freecad(+0x602a) [0x624e6be2b02a]
```

AjinkyaDahale, 2025-04-03:
@leoheck #20150 is meant to fix this issue. If you're able to compile please confirm.

leoheck, 2025-04-03:
By the way, this does not happen with the previous released weekly `FreeCAD 1.1.0, Libs: 1.1.0devR40971 (Git)`

buteomont, 2025-04-05:
I confirm that this fails in FreeCAD_weekly-builds-**41127**-conda-Linux-x86_64-py311.AppImage. The program crashes and force-closes when trimming a slot and a circle to make a keyhole.  It does _not_ crash on version FreeCAD_weekly-builds-**40504**-conda-Linux-x86_64-py311.AppImage.

AjinkyaDahale, 2025-04-05:
@buteomont as far as I know that buold doesn't contain the fix yet. Please compile from #20150.

RandyHut, 2025-04-06:
I just had the issue with version 41138

AjinkyaDahale, 2025-04-06:
@RandyHut 1.0 will do just fine instead of 0.21.2.

Please refrain from adding any more "doesn't work in 4xxxx" until and unless #20150 is merged.

macdroid53, 2025-04-06:
Additional comments: #810875

leoheck, 2025-04-10:
The fix did not work.
FreeCad is still crashing with the latest build: `FreeCAD 1.1.0, Libs: 1.1.0devR41213 (Git)`

See: 

https://github.com/user-attachments/assets/6feaecb9-7e59-4616-92d2-5f49499a589d

Can we reopen this issue?

```
➜ freecad_weekly_41213 /home/lheck/Downloads/trim_crash.FCStd
FreeCAD 1.1.0, Libs: 1.1.0devR41213 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

OSHAutoDoc workbench (v0.2.3) loaded
Initializing Assembly4 workbench (0.50.19-dev) ............................. done.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x78a72a245330]
#1  0x78a6316dac67 in Sketcher::SketchObject::isClosedCurve(Part::Geometry const*) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/lib/Sketcher.so+0x17
#2  0x78a6317276d3 in Sketcher::SketchObject::trim(int, Base::Vector3<double> const&) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/lib/Sketcher.so+0x753
#3  0x78a6317f7e33 in Sketcher::SketchObjectPy::trim(_object*) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/lib/Sketcher.so+0x93
#4  0x78a6317f80a4 in Sketcher::SketchObjectPy::staticCallback_trim(_object*, _object*) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/lib/Sketcher.so+0x24
#5  /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x1fae88) [0x78a72cbfae88]
#6  /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x2b4) [0x78a72cbd9b84]
#7  /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6e4) [0x78a72cbe5314]
#8  /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x29ee6d) [0x78a72cc9ee6d]
#9  /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x78a72cc9e864]
#10  /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x2bce33) [0x78a72ccbce33]
#11  /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x2b8a4a) [0x78a72ccb8a4a]
#12  /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x78a72ccacf29]
#13  0x78a72bf50058 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/libFreeCADBase.so+0x68
#14  0x78a72d5f3984 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0x94
#15  0x78a72d5f3b0f in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0xdf
#16  0x78a62c5f6016 in void Gui::cmdAppObjectArgs<int&, double&, double&>(App::DocumentObject const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&, double&, double&) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/lib/./SketcherGui.so+0x186
#17  /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/lib/./SketcherGui.so(+0x1f62cd) [0x78a62c5f62cd]
#18  0x78a62c74d1cc in SketcherGui::ViewProviderSketch::mouseMove(SbVec2s const&, Gui::View3DInventorViewer*) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/lib/./SketcherGui.so+0x4dc
#19  0x78a72da0e830 in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0x370
#20  0x78a729cd346e in SoEventCallback::handleEvent(SoHandleEventAction*) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/./libCoin.so.80+0xbe
#21  0x78a729ad54e8 in SoAction::traverse(SoNode*) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x68
#22  0x78a729c8cb9a in SoChildList::traverse(SoAction*, int, int) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/./libCoin.so.80+0xea
#23  0x78a729cdc348 in SoGroup::doAction(SoAction*) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x58
#24  0x78a729cf713a in SoSeparator::doAction(SoAction*) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x2a
#25  0x78a729ad54e8 in SoAction::traverse(SoNode*) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x68
#26  0x78a729c8cb9a in SoChildList::traverse(SoAction*, int, int) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/./libCoin.so.80+0xea
#27  0x78a729cdc348 in SoGroup::doAction(SoAction*) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x58
#28  0x78a729cf713a in SoSeparator::doAction(SoAction*) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x2a
#29  0x78a729ad54e8 in SoAction::traverse(SoNode*) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x68
#30  0x78a729adf26c in SoHandleEventAction::beginTraversal(SoNode*) from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/./libCoin.so.80+0xac
[... cut ...]
#60  0x78a72d4fa124 in Gui::Application::runApplication() from /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0x734
#61  /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/freecad(+0x5ca0) [0x595d13dd8ca0]
#62  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x78a72a22a1ca]
#63  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x78a72a22a28b]
#64  /home/lheck/Applications/FreeCAD-41213/squashfs-root/usr/bin/freecad(+0x602a) [0x595d13dd902a]
```

AjinkyaDahale, 2025-04-10:
> [@hyarion](https://github.com/hyarion) can we reopen this issue?

Please wait for the next builds. See https://github.com/FreeCAD/FreeCAD/issues/19770#issuecomment-2792169881.

RandyHut, 2025-04-10:
> Please wait for the next builds. See #19770 (comment)

I was instructed to not post, doesn't work in xxx so I have no comment.
Just watching to see when/if it is corrected.

AjinkyaDahale, 2025-04-10:
> I was instructed to not post, doesn't work in xxx so I have no comment.
> Just watching to see when/if it is corrected.
> […](#)

That instruction was only until #20150 is merged, which it is. Let me know if it is corrected after 41223.

Sorry if I came too strong at that time, I was annoyed that the fix, which has been ready for many weeks, hadn't been merged and I was getting the same updates, which, while well meaning, did not add any more information.

It does seem that some more work needs to be done to make it work with more configs (that I do not have), so an update from the later version will give me more info to fix.

hyarion, 2025-04-10:
@AjinkyaDahale sorry it took so long, feel free to poke me next time it happens so we can be annoyed together :)

RandyHut, 2025-04-13:
Just blew up when i tried to trim a 1" circle inside a  1" square --
Crashed on the first trim - latest conda file dropped yesterday

Randy

leoheck, 2025-04-13:
I saw it too. This is still happening on `1.1.0devR41232`

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x74eb24845330]
#1  0x74ea2bedab37 in Sketcher::SketchObject::isClosedCurve(Part::Geometry const*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/lib/Sketcher.so+0x17
#2  0x74ea2bf1f4ac in Sketcher::SketchObject::trim(int, Base::Vector3<double> const&) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/lib/Sketcher.so+0x59c
#3  0x74ea2bff7aa3 in Sketcher::SketchObjectPy::trim(_object*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/lib/Sketcher.so+0x93
#4  0x74ea2bff7d14 in Sketcher::SketchObjectPy::staticCallback_trim(_object*, _object*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/lib/Sketcher.so+0x24
#5  /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb048) [0x74eb271fb048]
#6  /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x2b4) [0x74eb271d9cf4]
#7  /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6e4) [0x74eb271e5494]
#8  /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x29ee5d) [0x74eb2729ee5d]
#9  /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x74eb2729e854]
#10  /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x2bce13) [0x74eb272bce13]
#11  /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x2b8a1a) [0x74eb272b8a1a]
#12  /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x74eb272acf19]
#13  0x74eb26550058 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libFreeCADBase.so+0x68
#14  0x74eb27bf3984 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0x94
#15  0x74eb27bf3b0f in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0xdf
#16  0x74ea229f19a6 in void Gui::cmdAppObjectArgs<int&, double&, double&>(App::DocumentObject const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&, double&, double&) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/lib/./SketcherGui.so+0x186
#17  /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/lib/./SketcherGui.so(+0x1f1c6e) [0x74ea229f1c6e]
#18  /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/lib/./SketcherGui.so(+0x1f1dad) [0x74ea229f1dad]
#19  0x74ea22b5cf96 in SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/lib/./SketcherGui.so+0x626
#20  0x74eb2800e622 in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0x162
#21  0x74eb242d346e in SoEventCallback::handleEvent(SoHandleEventAction*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libCoin.so.80+0xbe
#22  0x74eb240d54e8 in SoAction::traverse(SoNode*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x68
#23  0x74eb2428cb9a in SoChildList::traverse(SoAction*, int, int) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libCoin.so.80+0xea
#24  0x74eb242dc348 in SoGroup::doAction(SoAction*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x58
#25  0x74eb242f713a in SoSeparator::doAction(SoAction*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x2a
#26  0x74eb240d54e8 in SoAction::traverse(SoNode*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x68
#27  0x74eb2428cb9a in SoChildList::traverse(SoAction*, int, int) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libCoin.so.80+0xea
#28  0x74eb242dc348 in SoGroup::doAction(SoAction*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x58
#29  0x74eb242f713a in SoSeparator::doAction(SoAction*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x2a
#30  0x74eb240d54e8 in SoAction::traverse(SoNode*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x68
#31  0x74eb240df26c in SoHandleEventAction::beginTraversal(SoNode*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libCoin.so.80+0xac
#32  0x74eb240d61ba in SoAction::apply(SoNode*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x2ea
#33  0x74eb242a8ad2 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libCoin.so.80+0x72
[... cut ...]
#61  0x74eb27afa124 in Gui::Application::runApplication() from /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0x734
#62  /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/freecad(+0x5ca0) [0x644b2664aca0]
#63  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x74eb2482a1ca]
#64  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x74eb2482a28b]
#65  /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/freecad(+0x602a) [0x644b2664b02a]
```

leoheck, 2025-04-13:
https://github.com/user-attachments/assets/b2c13776-8f9b-4597-b11a-362f465bd4f2

AjinkyaDahale, 2025-04-13:
Still not able to reproduce with
```
OS: Manjaro Linux (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.41232 (Git) Conda AppImage
Build date: 2025/04/11 14:31:02
Build type: Release
Branch: main
Hash: 1983052e381b5e5dbc78778186dc8f85a23163e5
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Curves 0.6.47
  * Curves.backup1662220878.4365516 0.4.4
  * ExtremeProDark 2.5.3
  * ProDarkThemePreferencePack 1.0.0
  * ProDarkThemePreferencePack.backup1662220881.5889003 1.0.0
```

macdroid53, 2025-04-13:
I have not had a problem on Manjaro for a week or so no. Note: I use this version of the PKGBUILD since the official AUR freecad-git was failing to build for some time and this has been working ( https://github.com/random-cadder/freecad-aur )

I only have Debian machines where it is an issue ATM.

macdroid53, 2025-04-13:
I just build today's code on the Debian machine that crashes when the Trim is attempted. It still crashes.

leoheck, 2025-04-13:
We could compare libs this way
This is version 41232.

```
~/Applications/FreeCAD-41232 …
➜ ldd squashfs-root/usr/bin/freecad  
	linux-vdso.so.1 (0x00007ffc32b42000)
	libFreeCADGui.so => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libFreeCADGui.so (0x00007bbab1600000)
	libpython3.11.so.1.0 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0 (0x00007bbab1000000)
	libFreeCADApp.so => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libFreeCADApp.so (0x00007bbab0800000)
	libFreeCADBase.so => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libFreeCADBase.so (0x00007bbab0400000)
	libQt6Widgets.so.6 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libQt6Widgets.so.6 (0x00007bbaafc00000)
	libQt6Gui.so.6 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libQt6Gui.so.6 (0x00007bbaaf200000)
	libQt6Core.so.6 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libQt6Core.so.6 (0x00007bbaaec00000)
	libstdc++.so.6 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libstdc++.so.6 (0x00007bbaaea1e000)
	libgcc_s.so.1 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/libgcc_s.so.1 (0x00007bbab3b66000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007bbaae800000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007bbab3b37000)
	libGL.so.1 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libGL.so.1 (0x00007bbab3aac000)
	libCoin.so.80 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libCoin.so.80 (0x00007bbaade00000)
	libQt6PrintSupport.so.6 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libQt6PrintSupport.so.6 (0x00007bbab0f86000)
	libQt6SvgWidgets.so.6 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libQt6SvgWidgets.so.6 (0x00007bbab3a9e000)
	libQt6Network.so.6 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libQt6Network.so.6 (0x00007bbaadc2f000)
	libQt6UiTools.so.6 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libQt6UiTools.so.6 (0x00007bbab0f00000)
	libspnav.so.0 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libspnav.so.0 (0x00007bbab3a93000)
	libpyside6.cpython-311-x86_64-linux-gnu.so.6.7 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libpyside6.cpython-311-x86_64-linux-gnu.so.6.7 (0x00007bbab3a57000)
	libxerces-c-3.2.so => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libxerces-c-3.2.so (0x00007bbaad800000)
	libQt6Xml.so.6 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libQt6Xml.so.6 (0x00007bbab15d8000)
	libQt6Svg.so.6 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libQt6Svg.so.6 (0x00007bbab0e81000)
	libQt6OpenGLWidgets.so.6 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libQt6OpenGLWidgets.so.6 (0x00007bbab15c9000)
	libQt6OpenGL.so.6 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libQt6OpenGL.so.6 (0x00007bbab075a000)
	libshiboken6.cpython-311-x86_64-linux-gnu.so.6.7 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libshiboken6.cpython-311-x86_64-linux-gnu.so.6.7 (0x00007bbab06e0000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007bbaafb17000)
	/lib64/ld-linux-x86-64.so.2 (0x00007bbab3ba7000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007bbab3a4e000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007bbab15c4000)
	libboost_program_options.so.1.86.0 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libboost_program_options.so.1.86.0 (0x00007bbab03a1000)
	libz.so.1 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libz.so.1 (0x00007bbab06c5000)
	libEGL.so.1 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libEGL.so.1 (0x00007bbab038e000)
	libfontconfig.so.1 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libfontconfig.so.1 (0x00007bbaafaca000)
	libX11.so.6 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libX11.so.6 (0x00007bbaad6b7000)
	libglib-2.0.so.0 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libglib-2.0.so.0 (0x00007bbaad564000)
	libQt6DBus.so.6 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libQt6DBus.so.6 (0x00007bbaae740000)
	libxkbcommon.so.0 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libxkbcommon.so.0 (0x00007bbaadbe3000)
	libpng16.so.16 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libpng16.so.16 (0x00007bbaae703000)
	libharfbuzz.so.0 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libharfbuzz.so.0 (0x00007bbaad439000)
	libfreetype.so.6 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libfreetype.so.6 (0x00007bbaad37c000)
	libicui18n.so.75 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libicui18n.so.75 (0x00007bbaace00000)
	libicuuc.so.75 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libicuuc.so.75 (0x00007bbaaca00000)
	libdouble-conversion.so.3 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./libdouble-conversion.so.3 (0x00007bbab037d000)
[... cut ...]
	libdbus-1.so.3 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/././libdbus-1.so.3 (0x00007bbaaa5ab000)
	libgraphite2.so.3 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/././libgraphite2.so.3 (0x00007bbaacc37000)
	libXau.so.6 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./././libXau.so.6 (0x00007bbab06be000)
	libXdmcp.so.6 => /home/lheck/Applications/FreeCAD-41232/squashfs-root/usr/bin/../lib/./././libXdmcp.so.6 (0x00007bbaaf1f8000)
```

leoheck, 2025-04-13:
These are the ones that are not inside the AppImage

```bash
➜ ldd squashfs-root/usr/bin/freecad | grep -v home
	linux-vdso.so.1 (0x00007ffdaf9ac000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x000079cc4c600000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x000079cc5195a000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x000079cc4d917000)
	/lib64/ld-linux-x86-64.so.2 (0x000079cc519ca000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x000079cc4f3bc000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x000079cc4ecab000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x000079cc4ec78000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x000079cc4e18e000)
```

EchoJunction, 2025-04-14:
This is still happening in version 41232 on Linux.

Here is the crashlog: 
```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x42200) [0x76f494842200]
#1  0x76f415edab37 in Sketcher::SketchObject::isClosedCurve(Part::Geometry const*) from /tmp/.mount_FreeCAOpbapN/usr/lib/Sketcher.s
#2  0x76f415f1f4ac in Sketcher::SketchObject::trim(int, Base::Vector3<double> const&) from /tmp/.mount_FreeCAOpbapN/usr/lib/Sketche
#3  0x76f415ff7aa3 in Sketcher::SketchObjectPy::trim(_object*) from /tmp/.mount_FreeCAOpbapN/usr/lib/Sketcher.so+0x93
#4  0x76f415ff7d14 in Sketcher::SketchObjectPy::staticCallback_trim(_object*, _object*) from /tmp/.mount_FreeCAOpbapN/usr/lib/Sketc
#5  /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb048) [0x76f4971fb048]
#6  /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x2b4) [0x76f4971d9cf4]
#7  /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6e4) [0x76f4971e5494]
#8  /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/libpython3.11.so.1.0(+0x29ee5d) [0x76f49729ee5d]
#9  /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x76f49729e854]
#10  /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/libpython3.11.so.1.0(+0x2bce13) [0x76f4972bce13]
#11  /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/libpython3.11.so.1.0(+0x2b8a1a) [0x76f4972b8a1a]
#12  /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x76f4972acf19]
#13  0x76f496550058 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/l
#14  0x76f497bf3984 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAOp
#15  0x76f497bf3b0f in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_Free
#16  0x76f3821f19a6 in void Gui::cmdAppObjectArgs<int&, double&, double&>(App::DocumentObject const*, std::__cxx11::basic_string<ch
#17  /tmp/.mount_FreeCAOpbapN/usr/lib/./SketcherGui.so(+0x1f1c6e) [0x76f3821f1c6e]
#18  /tmp/.mount_FreeCAOpbapN/usr/lib/./SketcherGui.so(+0x1f1dad) [0x76f3821f1dad]
#19  0x76f38235cf96 in SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer con
#20  0x76f49800e622 in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/libFr
#21  0x76f4942d346e in SoEventCallback::handleEvent(SoHandleEventAction*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/./libCoin.so
#22  0x76f4940d54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/./libCoin.so.80+0x68
#23  0x76f49428cb9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/./libCoin.so.80+0xea
#24  0x76f4942dc348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/./libCoin.so.80+0x58
#25  0x76f4942f713a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/./libCoin.so.80+0x2a
#26  0x76f4940d54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/./libCoin.so.80+0x68
#27  0x76f49428cb9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/./libCoin.so.80+0xea
#28  0x76f4942dc348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/./libCoin.so.80+0x58
#29  0x76f4942f713a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/./libCoin.so.80+0x2a
#30  0x76f4940d54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/./libCoin.so.80+0x68
#31  0x76f4940df26c in SoHandleEventAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/./libCoin.so.80+0x
#32  0x76f4940d61ba in SoAction::apply(SoNode*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/./libCoin.so.80+0x2ea
#33  0x76f4942a8ad2 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/./libCoin.
#34  0x76f497f8b76e in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAOpbapN/usr/bin/..
#35  0x76f497f97368 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAOpbapN/usr/b
#36  0x76f497ff0e00 in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/libFreeCAD
#37  0x76f497ff585e in Gui::CADNavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/libFree
#38  0x76f497ff2765 in Gui::NavigationStyle::processEvent(SoEvent const*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/libFreeCADGu
#39  0x76f497fbcd4b in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/libFr
#40  0x76f497f84d9d in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../l
#41  0x76f494f4cfd6 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAOpbapN/usr/
#42  0x76f4961a40e4 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/libQt6Wid
#43  0x76f4961a4dcf in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAOpbapN/usr/bin/../lib/libQt6Widgets.so.6+0xc1
[... cut ...]
Build date: 2025/04/11 14:31:02
Build type: Release
Branch: main
Hash: 1983052e381b5e5dbc78778186dc8f85a23163e5
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.1131/1
Installed mods:
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * CadbaseLibrary 2.0.0
  * Gridfinity 0.11.4
  * ThreadProfile 1.93.0
  * Curves 0.6.60
  * OpenTheme 2024.9.1
  * IconThemes
```

leoheck, 2025-04-14:
I tested this issue on some old weklies I have.

The problem was introduced between version `40971` and `41127`.

These are the AppImages I have that I could test this.

This issue does not happen on:

1. FreeCAD_weekly-builds-36438-2024-03-15-conda-Linux-x86_64-py310.AppImage
2. FreeCAD_weekly-builds-38553-conda-Linux-x86_64-py311.AppImage
3. FreeCAD_weekly-builds-40408-conda-Linux-x86_64-py311.AppImage
4. FreeCAD_weekly-builds-40444-conda-Linux-x86_64-py311.AppImage
5. FreeCAD_weekly-builds-40655-conda-Linux-x86_64-py311.AppImage
6. FreeCAD_weekly-builds-40971-conda-Linux-x86_64-py311.AppImage

It does happen here:

1. FreeCAD_weekly-builds-41127-conda-Linux-x86_64-py311.AppImage
2. FreeCAD_weekly-builds-41138-conda-Linux-x86_64-py311.AppImage
3. FreeCAD_weekly-builds-41213-conda-Linux-x86_64-py311.AppImage
4. FreeCAD_weekly-builds-41232-conda-Linux-x86_64-py311.AppImage

leoheck, 2025-04-14:
@AjinkyaDahale, My appimages are here if it is helpful for someone.

https://www.dropbox.com/scl/fo/kk8ffuxc15ioj5l2b9r32/ABUUCxxigx1nlgVQMaPhdG8?rlkey=60qkhu7fwaxlh0m5p6jeaeonx&dl=0

AjinkyaDahale, 2025-04-14:
@leoheck thanks for checking but I think that's not relevant right now. It is all but certain that this crash was added when I effectively rewrote the method and the changes were merged into main.

The crash happens because a certain pointer is invalid. My efforts were to avoid calling the method at all if that is the case, but @hyarion pointed out in a private communication that I should check for the validity before bothering with that. Hopefully I'm able to get to it in a couple days. It should be a 2 line fix.

eblanshey, 2025-04-19:
Unfortunately trimming still causes a crash for me on the latest weekly 41314, which was built after https://github.com/FreeCAD/FreeCAD/pull/20800 was merged.

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x74a5c6045330]
#1  0x74a49fad9857 in Sketcher::SketchObject::isClosedCurve(Part::Geometry const*) from /tmp/.mount_freecaibkdIP/usr/lib/Sketcher.so+0x17
#2  0x74a49fb1d5fc in Sketcher::SketchObject::trim(int, Base::Vector3<double> const&) from /tmp/.mount_freecaibkdIP/usr/lib/Sketcher.so+0x5ac
#3  0x74a49fbf44e3 in Sketcher::SketchObjectPy::trim(_object*) from /tmp/.mount_freecaibkdIP/usr/lib/Sketcher.so+0x93
#4  0x74a49fbf4754 in Sketcher::SketchObjectPy::staticCallback_trim(_object*, _object*) from /tmp/.mount_freecaibkdIP/usr/lib/Sketcher.so+0x24
#5  /tmp/.mount_freecaibkdIP/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb048) [0x74a5c89fb048]
#6  /tmp/.mount_freecaibkdIP/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x2b4) [0x74a5c89d9cf4]
#7  /tmp/.mount_freecaibkdIP/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6e4) [0x74a5c89e5494]
#8  /tmp/.mount_freecaibkdIP/usr/bin/../lib/libpython3.11.so.1.0(+0x29ee5d) [0x74a5c8a9ee5d]
#9  /tmp/.mount_freecaibkdIP/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x74a5c8a9e854]
#10  /tmp/.mount_freecaibkdIP/usr/bin/../lib/libpython3.11.so.1.0(+0x2bce13) [0x74a5c8abce13]
#11  /tmp/.mount_freecaibkdIP/usr/bin/../lib/libpython3.11.so.1.0(+0x2b8a1a) [0x74a5c8ab8a1a]
#12  /tmp/.mount_freecaibkdIP/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x74a5c8aacf19]
#13  0x74a5c7d4ffe8 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_freecaibkdIP/usr/bin/../lib/libFreeCADBase.so+0x68
#14  0x74a5c93f4c34 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_freecaibkdIP/usr/bin/../lib/libFreeCADGui.so+0x94
#15  0x74a5c93f4dbf in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_freecaibkdIP/usr/bin/../lib/libFreeCADGui.so+0xdf
#16  0x74a49edf1886 in void Gui::cmdAppObjectArgs<int&, double&, double&>(App::DocumentObject const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&, double&, double&) from /tmp/.mount_freecaibkdIP/usr/lib/./SketcherGui.so+0x186
#17  /tmp/.mount_freecaibkdIP/usr/lib/./SketcherGui.so(+0x1f1b4e) [0x74a49edf1b4e]
#18  /tmp/.mount_freecaibkdIP/usr/lib/./SketcherGui.so(+0x1f1c8d) [0x74a49edf1c8d]
#19  0x74a49ef5cc96 in SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) from /tmp/.mount_freecaibkdIP/usr/lib/./SketcherGui.so+0x626
```

Version:
```
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41314 (Git) AppImage
Build date: 2025/04/18 15:56:21
Build type: Release
Branch: main
Hash: 2a983b587d374f584039797a1846bbb86753593d
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/140.99/1
Installed mods: 
  * WebTools 1.0.0
  * Curves 0.6.58
  * Plot 2024.11.26
  * AirPlaneDesign 0.4.1
  * Help 1.0.3
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
```

leoheck, 2025-04-20:
Unfortunately I am getting the same results.

@AjinkyaDahale does this backtrace using gdb helps investigating this issue?

```c++
Thread 1 "freecad" received signal SIGSEGV, Segmentation fault.
0x00007fff040d9857 in Sketcher::SketchObject::isClosedCurve(Part::Geometry const*) () from /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so
(gdb) bt
#0  0x00007fff040d9857 in Sketcher::SketchObject::isClosedCurve(Part::Geometry const*) () from /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so
#1  0x00007fff0411d5fc in Sketcher::SketchObject::trim(int, Base::Vector3<double> const&) () from /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so
#2  0x00007fff041f44e3 in Sketcher::SketchObjectPy::trim(_object*) () from /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so
#3  0x00007fff041f4754 in Sketcher::SketchObjectPy::staticCallback_trim(_object*, _object*) () from /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so
#4  0x00007ffff55fb048 in cfunction_call (func=0x7fffd5df2e80, args=<optimized out>, kwargs=<optimized out>) at /usr/local/src/conda/python-3.11.12/Objects/methodobject.c:553
#5  0x00007ffff55d9cf4 in _PyObject_MakeTpCall (tstate=0x7ffff59759b8 <_PyRuntime+166328>, callable=0x7fffd5df2e80, args=<optimized out>, nargs=2, keywords=0x0)
    at /usr/local/src/conda/python-3.11.12/Objects/call.c:214
#6  0x00007ffff55e5494 in _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=<optimized out>, throwflag=<optimized out>) at /usr/local/src/conda/python-3.11.12/Python/ceval.c:4769
#7  0x00007ffff569ee5d in _PyEval_EvalFrame (throwflag=0, frame=0x7ffff1fa3020, tstate=0x7ffff59759b8 <_PyRuntime+166328>)
    at /usr/local/src/conda/python-3.11.12/Include/internal/pycore_ceval.h:73
#8  _PyEval_Vector (tstate=tstate@entry=0x7ffff59759b8 <_PyRuntime+166328>, func=func@entry=0x7fff013b2980, locals=locals@entry=0x7fffee3f4d80, args=args@entry=0x0, 
    argcount=argcount@entry=0, kwnames=kwnames@entry=0x0) at /usr/local/src/conda/python-3.11.12/Python/ceval.c:6434
#9  0x00007ffff569e854 in PyEval_EvalCode (co=0x7fffd5f02ad0, globals=<optimized out>, locals=0x7fffee3f4d80) at /usr/local/src/conda/python-3.11.12/Python/ceval.c:1148
#10 0x00007ffff56bce13 in run_eval_code_obj (tstate=tstate@entry=0x7ffff59759b8 <_PyRuntime+166328>, co=co@entry=0x7fffd5f02ad0, globals=globals@entry=0x7fffee3f4d80, 
    locals=locals@entry=0x7fffee3f4d80) at /usr/local/src/conda/python-3.11.12/Python/pythonrun.c:1741
#11 0x00007ffff56b8a1a in run_mod (mod=mod@entry=0x55555dea18a0, filename=filename@entry=0x7ffff5952510 <_PyRuntime+21776>, globals=globals@entry=0x7fffee3f4d80, 
    locals=locals@entry=0x7fffee3f4d80, flags=flags@entry=0x0, arena=arena@entry=0x7fff0378ff10) at /usr/local/src/conda/python-3.11.12/Python/pythonrun.c:1762
#12 0x00007ffff56acf19 in PyRun_StringFlags (str=<optimized out>, start=<optimized out>, globals=0x7fffee3f4d80, locals=0x7fffee3f4d80, flags=0x0)
    at /usr/local/src/conda/python-3.11.12/Python/pythonrun.c:1632
#13 0x00007ffff494ffe8 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) () from /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/bin/../lib/libFreeCADBase.so
#14 0x00007ffff5ff4c34 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) ()
   from /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/bin/../lib/libFreeCADGui.so
#15 0x00007ffff5ff4dbf in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) ()
   from /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/bin/../lib/libFreeCADGui.so
#16 0x00007ffef33f1886 in void Gui::cmdAppObjectArgs<int&, double&, double&>(App::DocumentObject const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&, double&, double&) () from /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/./SketcherGui.so
#17 0x00007ffef33f1b4e in SketcherGui::DrawSketchHandlerTrimming::executeCommands(Base::Vector2d) () from /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/./SketcherGui.so
#18 0x00007ffef33f1c8d in SketcherGui::DrawSketchHandlerTrimming::releaseButton(Base::Vector2d) () from /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/./SketcherGui.so
#19 0x00007ffef355cc96 in SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) ()
   from /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/./SketcherGui.so
#20 0x00007ffff640dfb2 in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) () from /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/bin/../lib/libFreeCADGui.so
#21 0x00007ffff26d346e in SoEventCallback::handleEvent(SoHandleEventAction*) () from /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/bin/../lib/./libCoin.so.80
#22 0x00007ffff24d54e8 in SoAction::traverse(SoNode*) () from /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/bin/../lib/./libCoin.so.80
#23 0x00007ffff268cb9a in SoChildList::traverse(SoAction*, int, int) () from /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/bin/../lib/./libCoin.so.80
#24 0x00007ffff26dc348 in SoGroup::doAction(SoAction*) () from /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/bin/../lib/./libCoin.so.80
[... cut ...]
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.73.0
  * Curves 0.6.60
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.3
  * lattice2 1.0.0
  * kicadStepUpMod 11.6.0
```

leoheck, 2025-04-20:
ChatGPT gave me some insight, it might give you some new ideas on how to find this little bug.

[image]

leoheck, 2025-04-20:
Interesting that I tried to run it with Valgrind. 

This way (which is not exactly the way it should run), since I am not using the `AppRun` script in this case.

```bash
valgrind --error-limit=no /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/bin/freecad trim_crash.FCStd-
```

And in this case, the trim worked fine.

I am not sure if this is useful, but these are the last lines of valgrind.
```
==26532== 
==26532== Invalid read of size 8
==26532==    at 0x57E3D652: Sketcher::SketchObject::trim(int, Base::Vector3<double> const&) (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so)
==26532==    by 0x57F144E2: Sketcher::SketchObjectPy::trim(_object*) (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so)
==26532==    by 0x57F14753: Sketcher::SketchObjectPy::staticCallback_trim(_object*, _object*) (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so)
==26532==    by 0x6E9F047: cfunction_call (methodobject.c:553)
==26532==    by 0x6E7DCF3: _PyObject_MakeTpCall (call.c:214)
==26532==    by 0x6E89493: _PyEval_EvalFrameDefault (ceval.c:4769)
==26532==    by 0x6F42E5C: UnknownInlinedFun (pycore_ceval.h:73)
==26532==    by 0x6F42E5C: _PyEval_Vector (ceval.c:6434)
==26532==    by 0x6F42853: PyEval_EvalCode (ceval.c:1148)
==26532==    by 0x6F60E12: run_eval_code_obj (pythonrun.c:1741)
==26532==    by 0x6F5CA19: run_mod (pythonrun.c:1762)
==26532==    by 0x6F50F18: PyRun_StringFlags (pythonrun.c:1632)
==26532==    by 0x7A2AFE7: Base::InterpreterSingleton::runString[abi:cxx11](char const*) (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/libFreeCADBase.so)
==26532==  Address 0xcf1a6020 is 0 bytes inside a block of size 56 free'd
==26532==    at 0x484A61D: operator delete(void*, unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==26532==    by 0x327218CE: Part::PropertyGeometryList::setValues(std::vector<Part::Geometry*, std::allocator<Part::Geometry*> >&&) (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Part.so)
==26532==    by 0x57E3348B: Sketcher::SketchObject::solve(bool) (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so)
==26532==    by 0x57E3A2A4: Sketcher::SketchObject::delConstraints(std::vector<int, std::allocator<int> >, bool) (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so)
==26532==    by 0x57E3D5D6: Sketcher::SketchObject::trim(int, Base::Vector3<double> const&) (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so)
==26532==    by 0x57F144E2: Sketcher::SketchObjectPy::trim(_object*) (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so)
==26532==    by 0x57F14753: Sketcher::SketchObjectPy::staticCallback_trim(_object*, _object*) (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so)
==26532==    by 0x6E9F047: cfunction_call (methodobject.c:553)
==26532==    by 0x6E7DCF3: _PyObject_MakeTpCall (call.c:214)
==26532==    by 0x6E89493: _PyEval_EvalFrameDefault (ceval.c:4769)
==26532==    by 0x6F42E5C: UnknownInlinedFun (pycore_ceval.h:73)
==26532==    by 0x6F42E5C: _PyEval_Vector (ceval.c:6434)
==26532==    by 0x6F42853: PyEval_EvalCode (ceval.c:1148)
==26532==  Block was alloc'd at
==26532==    at 0x4846FA3: operator new(unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==26532==    by 0x328CAD65: Part::GeomLineSegment::copy() const (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Part.so)
==26532==    by 0x328AB979: Part::Geometry::clone() const (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Part.so)
==26532==    by 0x57F4EF90: Sketcher::Sketch::extractGeometry(bool, bool) const (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so)
==26532==    by 0x57E33417: Sketcher::SketchObject::solve(bool) (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so)
==26532==    by 0x57E4E5E3: Sketcher::SketchObject::execute() (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so)
==26532==    by 0x74D65A1: App::DocumentObject::recompute() (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/libFreeCADApp.so)
==26532==    by 0x326EC96B: Part::Feature::recompute() (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Part.so)
==26532==    by 0x74B091B: App::Document::_recomputeFeature(App::DocumentObject*) (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/libFreeCADApp.so)
==26532==    by 0x74BA3E4: App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/libFreeCADApp.so)
==26532==    by 0x753B74B: App::DocumentPy::recompute(_object*) (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/libFreeCADApp.so)
==26532==    by 0x753B963: App::DocumentPy::staticCallback_recompute(_object*, _object*) (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/libFreeCADApp.so)
==26532== 
==26532== Invalid read of size 8
==26532==    at 0x57E3D683: Sketcher::SketchObject::trim(int, Base::Vector3<double> const&) (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so)
==26532==    by 0x57F144E2: Sketcher::SketchObjectPy::trim(_object*) (in /home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/Sketcher.so)
[... cut ...]
==26532==                         newarray           : 78,800 bytes in 523 blocks
==26532==                         multipleinheritance: 1,191,728 bytes in 6,934 blocks
==26532==         suppressed: 0 bytes in 0 blocks
==26532== Rerun with --leak-check=full to see details of leaked memory
==26532== 
==26532== Use --track-origins=yes to see where uninitialised values come from
==26532== For lists of detected and suppressed errors, rerun with: -s
==26532== ERROR SUMMARY: 2075 errors from 1519 contexts (suppressed: 0 from 0)

 ```

RandyHut, 2025-04-20:
I downloaded the 41364 file for Linux - running the ap-img file, it
completely crashes the program when i try to trim a circle or sq when they
are the same dim centered.

[CODE]System:
  Kernel: 6.12.20-amd64 [6.12.20-1~mx23ahs] arch: x86_64 bits: 64 compiler:
N/A
    parameters: BOOT_IMAGE=/boot/vmlinuz-6.12.20-amd64 root=UUID=<filter>
ro quiet splash
  Desktop: Xfce v: 4.20.0 tk: Gtk v: 3.24.38 info: xfce4-panel wm: xfwm v:
4.20.0 vt: 7
    dm: LightDM v: 1.32.0 Distro: MX-23.6_x64 Libretto September 15  2024
base: Debian GNU/Linux 12
    (bookworm)
Machine:
  Type: Desktop System: Hewlett-Packard product: HP Compaq Elite 8300 SFF
v: N/A
    serial: <superuser required> Chassis: type: 4 serial: <superuser
required>
  Mobo: Hewlett-Packard model: 3397 serial: <superuser required> UEFI:
Hewlett-Packard
    v: K01 v03.08 date: 04/10/2019
Battery:
  Device-1: hidpp_battery_0 model: Logitech Wireless Keyboard serial:
<filter>
    charge: 55% (should be ignored) rechargeable: yes status: discharging
CPU:
  Info: model: Intel Core i7-3770 bits: 64 type: MT MCP arch: Ivy Bridge
gen: core 3 level: v2
    built: 2012-15 process: Intel 22nm family: 6 model-id: 0x3A (58)
stepping: 9 microcode: 0x21
  Topology: cpus: 1x cores: 4 tpc: 2 threads: 8 smt: enabled cache: L1: 256
KiB
    desc: d-4x32 KiB; i-4x32 KiB L2: 1024 KiB desc: 4x256 KiB L3: 8 MiB
desc: 1x8 MiB
  Speed (MHz): avg: 1600 high: 1629 min/max: 1600/3900 scaling: driver:
intel_cpufreq
    governor: schedutil cores: 1: 1600 2: 1596 3: 1596 4: 1596 5: 1596 6:
1596 7: 1629 8: 1596
    bogomips: 54277
  Flags: avx ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
  Vulnerabilities:
  Type: gather_data_sampling status: Not affected
  Type: itlb_multihit status: KVM: Split huge pages
  Type: l1tf mitigation: PTE Inversion; VMX: conditional cache flushes, SMT
vulnerable
  Type: mds mitigation: Clear CPU buffers; SMT vulnerable
  Type: meltdown mitigation: PTI
  Type: mmio_stale_data status: Unknown: No mitigations
  Type: reg_file_data_sampling status: Not affected
  Type: retbleed status: Not affected
  Type: spec_rstack_overflow status: Not affected
  Type: spec_store_bypass mitigation: Speculative Store Bypass disabled via
prctl
  Type: spectre_v1 mitigation: usercopy/swapgs barriers and __user pointer
sanitization
  Type: spectre_v2 mitigation: Retpolines; IBPB: conditional; IBRS_FW;
STIBP: conditional; RSB
    filling; PBRSB-eIBRS: Not affected; BHI: Not affected
  Type: srbds status: Vulnerable: No microcode
  Type: tsx_async_abort status: Not affected
Graphics:
  Device-1: Intel IvyBridge GT2 [HD Graphics 4000] vendor: Hewlett-Packard
driver: i915 v: kernel
    arch: Gen-7 process: Intel 22nm built: 2012-13 ports: active: HDMI-A-1
empty: DP-1,VGA-1
    bus-ID: 00:02.0 chip-ID: 8086:0162 class-ID: 0300
  Display: x11 server: X.Org v: 1.21.1.7 compositor: xfwm v: 4.20.0 driver:
X:
    loaded: modesetting unloaded: fbdev,vesa dri: crocus gpu: i915
[... 62 lines cut ...]
    pkgs: 0 pm: flatpak pkgs: 6
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/brave-browser-release.list
    1: deb [arch=amd64
signed-by=/usr/share/keyrings/brave-browser-archive-keyring.gpg]
https://brave-browser-apt-release.s3.brave.com/ stable main
  Active apt repos in: /etc/apt/sources.list.d/debian-stable-updates.list
    1: deb http://deb.debian.org/debian bookworm-updates main contrib
non-free non-free-firmware
  Active apt repos in: /etc/apt/sources.list.d/debian.list
    1: deb http://deb.debian.org/debian bookworm main contrib non-free
non-free-firmware
    2: deb http://security.debian.org/debian-security bookworm-security
main contrib non-free non-free-firmware
  Active apt repos in: /etc/apt/sources.list.d/mx.list
    1: deb http://mirror.math.princeton.edu/pub/mxlinux/mx/repo/ bookworm
main non-free
  No active apt repos in: /etc/apt/sources.list.d/extrepo_librewolf.sources
Info:
  Processes: 257 Uptime: 2h 33m wakeups: 14 Memory: 11.56 GiB used: 2.78
GiB (24.1%) Init: SysVinit
  v: 3.06 runlevel: 5 default: graphical tool: systemctl Compilers: gcc:
12.2.0 alt: 12
  Client: shell wrapper v: 5.2.15-release inxi: 3.3.26
Boot Mode: UEFI[/CODE]

AjinkyaDahale, 2025-04-20:
I'm already checking for `geom == nullptr`. Is it not null but somehow invalidated?

leoheck, 2025-04-20:
@AjinkyaDahale did you see my last comment above where I said that running Freecad using the `squashfs-root/AppRun` script crashes the app when trimming while launching it directly with `squashfs-root/usr/bin/freecad` (but with valgrind) didn't make it crash?

I still have to check if running it without valgrind (and not using the AppRun script) can work. 

If this is the case, some internal library or something that is added in the AppRun script is causing this problem on Debian based systems.

RandyHut, 2025-04-20:
Don't know if it helps, but i downloaded the latest Manjaro os wrote it to
a usb, loaded from the usb and then ranit on the usb. I downloaded the
latest freecad apimg and ran it. No difference, it crashed as soon as i
tried to trim the circle, in a sq.. if there would be any files on the usb
that might help, let me know. i just turned it off and removed it once it
was off and rebooted on my normal machine. I also tried it on an install of
mint 20 and it failed there as well.

AjinkyaDahale, 2025-04-20:
@leoheck I didn't quite get you, but it just may be that I don't have much knowledge of these tools. Regardless, I should definitely fix it (if someone else doesn't beat me to it) even if the problem is not in my code.

leoheck, 2025-04-20:
@AjinkyaDahale recently, I started using the AppImage extracted into a folder.

When we extract it, the `squashfs-root` folder is generated with a script (`AppRun`) at the root. This script is run when we execute the AppImage file itself.

This script loads some environment variables before calling the `squashfs-root/usr/bin/freecad`.

Then, when I tried to run `valgrind` to check for memory leaks, I thought I could not use the `AppRun` script. And then I decided to use the executable directly. Then I tested the trim issue, and I was not able to reproduce it!

But testing the executable right now, without using Valgrind did not solve the issue. 

However, using `valgrind`, I was able to trim the Sketch without the crash, as you can see in the screenshot. 

[image]

_So in short, the operation crashes the tool, but using Valgrind, it did not happen._

Here is an explanation about alignment and `double` that is related to the statement above.
https://stackoverflow.com/questions/40601655/program-crashes-normally-but-not-with-valgrind

Now, when valgrind was initializing, it complained about this, which is also related to double (or long double)

[image]

Here follows the text from the screenshot above if you want to copy something.

```
/home/lheck/Applications/FreeCAD-41314/squashfs-root/usr/lib/python3.11/site-packages/numpy/_core/getlimits.py:551: UserWarning: Signature b'\x00\xd0\xcc\xcc\xcc\xcc\xcc\xcc\xfb\xbf\x00\x00\x00\x00\x00\x00' for <class 'numpy.longdouble'> does not match any known type: falling back to type probe function.
This warnings indicates broken support for the dtype!
  machar = _get_machar(dtype)
```

mfro0, 2025-04-23:
For me, this simple trimming attempt (try to trim protruding line endings) fails (Segmentation Violation) even with the latest build:

[image]

```
OS: Debian GNU/Linux trixie/sid (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.41391 (Git)
Build date: 2025/04/23 02:37:17
Build type: Unknown
Branch: main
Hash: 4ff9cb343e997a7626ea3f9b2177c782374ebcc0
Python 3.13.3, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.16
shiboken 5.15.16, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.6656/1
Installed mods: 
  * GrabBag
  * Curves 0.6.61
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * Silk 0.1.5
  * freecad.gears 1.3.0

```

closed 2025-04-28 by chennes

RandyHut, 2025-04-29:
> completed via #20951 <https://github.com/FreeCAD/FreeCAD/pull/20951>.

Latest file (2hrs ago) WORKED! Trimmed without failure.
Randy

mfro0, 2025-04-29:
Confirmed. Trimming works as desired. Thank you!

leoheck, 2025-04-29:
@AjinkyaDahale, thanks. Now it is working fine!

commit b6a069dc (2025-04-10): [Sketcher] Attempt to avoid double-trimming in some cases

Possibly fixes #19425 and/or #20081.