FreeCAD/FreeCAD #19429: Assembly workbench uses blue origin plain for distance constraint regardless of selection

EstebanFPV, 2025-02-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Selected green origin plane and face.
[image]
[image]

Selected blue origin plane and face. 
[image]
[image]

Selected red origin plane and face.
[image]
[image]

The constaint does succeed. it constrains the plane to the face, just no control over which plane. which is why i was having trouble using 2 planes and 2 faces, as it was trying to constrain the same origin plane to multiple faces. 

Constraining 2 origin planes together always uses the blue plane for both objects. 

All 3 examples return the same results. tested with last 3 versions of weekly build. i did not attempt on the main branch. windows 10 .7z

Face to cylinder tangent behaves strangely too, but that's another topic.

Thanks,
Tim

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Logical/physical DPI: 96/141.585
Installed mods: 
  * 3DfindIT 1.2.0
  * Assembly3 0.12.2
  * OpenTheme 2024.5.16
  * sheetmetal 0.7.3
  * sheetmetal.backup1735846653.2831378 0.4.15 (Disabled)
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-06:
@PaddleStroke FYI

EstebanFPV, 2025-02-06:
i will share an assembly file later this am

EstebanFPV, 2025-02-06:
[image]

excuse me?

EstebanFPV, 2025-02-06:
[assembly test.zip](https://github.com/user-attachments/files/18695583/assembly.test.zip)

PaddleStroke, 2025-03-07:
This is fixed by https://github.com/FreeCAD/FreeCAD/pull/20028

PaddleStroke, 2025-03-07:
> It is, BTW, quite confusing that the dragger is automatically displayed when you select an origin object.

Yes I agree. I created https://github.com/FreeCAD/FreeCAD/issues/20047 for it

Roy-043, 2025-03-07:
@PaddleStroke thanks for fixing this.

closed 2025-03-07 by Roy-043

TimBuildIt, 2025-03-12:
@Roy-043 @PaddleStroke  is this fixed in the weekly? it still does it for me

Roy-043, 2025-03-12:
If you are using Windows: the Windows weekly is stuck ATM.