FreeCAD/FreeCAD #19434: [TechDraw] Inserting View will maximize the Page window

steveklett, 2025-02-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When a TechDraw Page is not maximized (e.g., tiled with a 3D view) and a new View is inserted the TechDraw Page will be maximized.

Steps to Reproduce

1. Change to TechDraw Workbench 
2. Tile a 3D window and a TechDraw Page
[image]

2. Select your model in 3D view and click Insert View on the toolbar
3. The TechDraw Page will be maximized
[image]

[issue - techdraw insert forces maximize.FCStd.tar.gz](https://github.com/user-attachments/files/18679852/issue.-.techdraw.insert.forces.maximize.FCStd.tar.gz)

### Full version info

```shell
OS: Linux Mint 22 (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96.0135/108.888
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-06:
@WandererFan I can confirm.
Also, when closing FreeCAD for a blink of a second the TD page gets shown as full screen.

WandererFan, 2025-02-08:
problem is in ViewProviderPage::showMDIViewPage().  The call to showMaximized() should only apply to a newly created tab.  For existing tabs, replace that call with MainWindow::setActiveWindow(this).

WandererFan, 2025-02-08:
Showing the page on shut down is something else I think.

maxwxyz, 2025-03-01:
@WandererFan @yorikvanhavre this issue still happens in the latest weekly, when double clicking on a Draft workingpane proxy:

https://github.com/user-attachments/assets/df0dcf53-bbd3-4884-9708-ddb2a2f1a549

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40444 (Git) Conda
Build type: Release
Branch: main
Hash: 089605bb4db98668661bf56db3f5f5bb4f7f0d87
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/108.888
Installed mods: 
  * Cables 0.1.3
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.2
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.20
  * Silk 0.1.5

```

WandererFan, 2025-03-01:
What I see in the video is not this issue.  The flashing is something else.

maxwxyz, 2025-03-01:
The flashing is the TD Page going full screen for a flash. The window title is the page name.

WandererFan, 2025-03-01:
Does this happen without all those floating windows/overlays?

maxwxyz, 2025-03-01:
Yes, but it takes 4hrs to build :)

closed 2025-09-28 by luzpaz