FreeCAD/FreeCAD #19437: Strange render when exporting PDF in Freecad 1.1.0dev

wiltur, 2025-02-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Elements are filled in gray, even the empty spaces between them. 

This does not happen with stable version.

Exported version 1.0.0 (Stable)
[image]

Exported version 1.1.0dev
[image]

The pdf
[test-1.1.0dev.pdf](https://github.com/user-attachments/files/18685910/test-1.1.0dev.pdf)

### Full version info

```shell
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40006 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * BIM021 2021.12.0
  * parts_library
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-02-06:
What are your settings for transparent faces and face colour? It looks like you have transparent faces turned on and face colour set to grey.

[image]

wiltur, 2025-02-06:
Well, it's the same if transparent faces is turned ON or OFF. (which I don't think is normal)
The only working solution seams to make the color white.

Thanks for pointing me in the right direction.

WandererFan, 2025-02-07:
@wiltur Does the page look right in FreeCAD?

wiltur, 2025-02-07:
Yes it's ok in Freecad, it looks "white" but not in the PDF.

also switching ON or OFF the transparency doesn't affect the result in the PDF (but it's ok for the screen render)

jsonmaur, 2025-06-07:
This still seems to be an issue in v1.0.1 (stable).

jsonmaur, 2025-06-07:
Figured it out! You have to select the view from the Tree, and set the face color back to white in the View tab.

IJMardon, 2025-11-27:
I have just encountered this issue and found the jsonmaur's Jun 7 fix of manually changing the Face Color of the View in Tree to work.  Any plans to update future versions to tie the View Face Color to the global TechDraw Preferences?

Version            1.0.0
Rev. Number   39109 (Git)
Release date   2024/11/18
OS                   Windows 11
Architecture    x86_64

FEA-eng, 2026-08-27:
> [@FEA-eng](https://github.com/FEA-eng) - I think this can be closed. There can still be an issue if an old face/page color preference is migrated to 1.1.1+. If the prefs are adjusted, then everything works OK.

@maxwxyz Can you close it then?

closed 2026-08-27 by maxwxyz