FreeCAD/FreeCAD #19440: Free Distance measurements are stacked and cannot be moved.

leoheck, 2025-02-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When measuring "Distance Free," where the Delta X, Y, and Z are displayed, I can move the Straight measure (the one with the black background), but I cannot move the ones with the white background, which are generally stacked on short distance measures.

[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.7.5
  * Curves 0.6.54
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-20:
@hlorus any chance you can take a look at this one ?

hlorus, 2025-05-20:
It is actually expected behavior that the delta labels are not movable. IMO it would get too confusing if all labels could be dragged around. Some alternatives could be:
- Allow user to show/hide individual components
- Auto hide components that are equal to 0
- Display the component values in the task dialog

leoheck, 2025-05-20:
Anything is better than stacked values that we cannot read.

Initially, your suggestion to display if the components in the task dialog would help a lot, and you don't even need to remove them from the screen. just giving an alternative place to read the values would be nice.

hlorus, 2025-05-20:
Just to be clear, the component values can always be seen in the data tab inside the property viewer:

[image]

leoheck, 2025-05-20:
Hm, didn't know that. But on the property view this kinda sucks.

Because then you have 3 palaces to display the information. Task dialog, 3d model and property view.

Also, the way this measure tool works is kinda annoying. It creates objects in the tree. I should have used this less than 1% of the time. Most of the time I have to quick measure things (measure and get the value to use somewhere else).

FEA-eng, 2025-05-30:
New forum thread about this: https://forum.freecad.org/viewtopic.php?t=97421

It would indeed be great to make the delta labels movable like the main distance label. And/or at least position them better initially.

alfadir, 2025-05-30:
Hi, I am sorry not to have noticed the issue, I tried to find a solution. 
One simple solution would be to add the measurements to the panel ? where the red lines are..  or maybe that is complex as well.

[image]

steveklett, 2026-02-06:
I struggle with this as well. Moving all 3 would be better than nothing, but also not ideal as it's complicated and the screen could get crazy. How about if they were a single, stacked widget. The colored reference lines would still be drawn and the measurements displayed in a stacked, single object that could be moved. For that matter, maybe combine all 4 together. Also the ability to show/hide each from the properties panel.

leoheck, 2026-02-10:
The information is indeed better with "Distance", dx, dy and dz

FEA-eng, 2026-02-12:
Fusion 360:

[image]

Creo:

[image]

JustTryingToGetSomeWorkDone, 2026-09-10:
Personally I do not think it is confusing at all to be able to move the labels. It is far more confusing that they are present and cannot be seen. I agree with previous comments that just removing them is less good than fixing it so they can be seen. Even if you consider everyone a caveman and moving is too much for our primitive minds to comprehend, space them out a little so we can read each one. It helps to learn that I can find those deltas in the details pane, but it was not immediately obvious to me because the labels are different. I was looking for deltas, not just a list of distances. That inconsistency probably needs a touch as well.

bkmeneguello, 2026-09-13:
I ran into this issue today and did a quick analysis to trace where it originates:

- In [`src/Mod/Measure/Gui/ViewProviderMeasureDistance.cpp#ViewProviderMeasureDistance::ViewProviderMeasureDistance`](https://github.com/FreeCAD/FreeCAD/blob/e273f684f85ba2b42214fc6b831e5a106c0cec8a/src/Mod/Measure/Gui/ViewProviderMeasureDistance.cpp#L421-L440), the deltas are split into three separate `MeasureGui::DimensionLinear` instances.
- In [`src/Mod/Measure/Gui/ViewProviderMeasureDistance.cpp#MeasureGui::DimensionLinear::setupDimension`](https://github.com/FreeCAD/FreeCAD/blob/e273f684f85ba2b42214fc6b831e5a106c0cec8a/src/Mod/Measure/Gui/ViewProviderMeasureDistance.cpp#L210), each label position is calculated simply by placing the label origin at the axis center offset by a fixed distance.

Key takeaways:
- There is no dedicated "Triple Axis Measure" component to intelligently arrange delta dimensions relative to one another.
- `MeasureGui::DimensionLinear` does not perform collision detection to prevent overlapping labels, nor does it support manual repositioning.

I don't have a patch ready, but hopefully this pointer saves some time for whoever tackles this!

oxfordpelican, 2026-09-29:
I've opened #33026 for the overlap part of this.

It doesn't make the labels movable. A delta label that would cover another one
just moves down until it's clear, including under the main distance label.

@bkmeneguello that's the collision detection you mentioned.

Branch is `measure-delta-label-overlap` on
https://github.com/oxfordpelican/FreeCAD if anyone wants to try it. Linux only
so far, could someone test it on Windows or macOS?