FreeCAD/FreeCAD #19441: Perpetual warning about wrong unit schema re CAM workbench, even with CNC schema is selected.

Bazmundi, 2025-02-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Apparently a PR #6367 "fixed" a problem where gcode was generated incorrectly since the problem was apparently wrong class of unit schema.  So there is a supposedly conditional warning dialog when you go to create a job. Very good.

But.

I am running CAM workbench for the first time (ever).

I got the warning which was informative so I did go to peferences to change the schema to "Metric, Small Parts & CNC".  Yet, when I go back to create a job, I still get the warning.  

I "applied" the schema change.  Exited FreeCAD and restarted and checked the unit schema, which was still set at "Metric, Small Parts & CNC".  

The threads I have found state emphatically that the warning is only supposed to appear if the unit schema is not one of the appropriate types.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Manipulator 1.5.7
  * Design456 0.0.1
  * FeedsAndSpeeds 0.6.0
  * LCInterlocking
  * Assembly4 0.50.16-dev
  * lattice2 1.0.0
  * fasteners 0.5.37
  * PieMenu 1.2.7
  * Curves 0.6.55
  * ThreadProfile 1.92.0
  * parts_library
  * freecad.gears 1.3.0
  * A2plus.backup1728897810.7832332 0.4.61 (Disabled)
  * Nodes 0.1.36
  * kicadStepUpMod 11.5.2
  * Beltrami 1.1.0
  * ImportNURBS 1.1.0
  * sheetmetal 0.7.5
  * A2plus 0.4.68
  * Gridfinity 0.8.3
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-12:
@jbaehr FYI

tarman3, 2025-02-12:
You using very old pre-release version of FreeCAD
Try last weekly build
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

jbaehr, 2025-02-16:
I think there is only a misunderstanding, not a bug: The unit scheme is now *per document*. This means, the setting in the preference is the default for new documents, but it does not change existing documents.
Please check the current project's unit settings: Menu bar: File -> Entry: Document Information -> Property: Unit System.

 So in some sense it's a duplicate of #11839, in response to which the check was moved to the job creation, now reading the document's setting (instead of the global one) and the text got improved: #11851.

Still, I think there is room for improvement. While we now have a the information "for this document" in there, it still advices to change the "preferences".

> The currently selected unit schema: 
>     '{}' **for this document**
> Does not use 'minutes' for velocity values. 
> 
> CNC machines require feed rate to be expressed in 
> unit/minute. To ensure correct G-code: 
> Select a minute-based schema **in preferences**.
> For example:
>     'Metric, Small Parts & CNC'
>     'US Customary'
>     'Imperial Decimal'

(emphasis by me, source: [JobDlg.py]( https://github.com/FreeCAD/FreeCAD/blob/04d2bcbf45ab96f3895ac3d0e8b5ae9becbaf5cd/src/Mod/CAM/Path/Main/Gui/JobDlg.py#L68))

I'll create a PR.

closed 2025-03-17 by chennes