FreeCAD/FreeCAD #19442: [Part] Improve FreeCAD Hidden Line Removal

yorikvanhavre, 2025-02-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Exported from https://github.com/opencax/GSoC/issues/69#issuecomment-2045951880

## Outline
FreeCAD's Technical Drawing module (TechDraw) relies heavily on the OpenCascade Hidden Line Removal algorithms.  These algorithms can be very slow, do not provide progress reporting and do not provide any linkage between the input shape and the output.

## Details
The TechDraw module provides projections, section views and detail views of 3D model components and assemblies developed in FreeCAD modules such as Part, PartDesign and Draft.

## Expected Outcome
a) develop new code for projecting shapes and creating the geometry for technical drawings. 
-or-
b) modify the existing OpenCascade code as an enhancement.

## Project Properties
Both OpenCascade and TechDraw are written in C++.

### Skills
The student should have a good knowledge of C++ and be familar with graphics topics such as the painters algorithm, face detection and hidden line removal.
Knowledge of technical drawing standards and previous exposure to Qt will be helpful.  Familiarity with OpenCascade is a definite plus.

### Difficulty 
Hard

### Size 
long

### Additional Information

*  Potential mentor(s): wandererfan
*  Organization website: https://freecadweb.org
*  Communication channels: https://forum.freecadweb.org

### Full version info

```shell
n/a
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-02-07:
IMO, this is too big a job for a GSOC project.  I would happy to be proven wrong.

yorikvanhavre, 2025-02-14:
Obviously, it's like Toponaming, it's a very large subject. But we had GSoC projects working on Toponaming already in the past, and even if they didn't reach a "workable state", it helped a lot to advance and ultimately reach a solution.

This is a very pure, theoretical, algorithmic project, I still think it would make a beautiful GSoC project for someone interested in exploring that part.