FreeCAD/FreeCAD #19443: Recorded macro failed to payback with redondant datum created.

mgth, 2025-02-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I Recorded a very simple macro to create a sketch containing a square of 20x20mm
In the resulting macro datum are created multiple times and the macro crash on replay.

[TestMacro2.zip](https://github.com/user-attachments/files/18691444/TestMacro2.zip)

In the zip you can find the file created while recording the macro and the recorded macro.
I got many other bugs while recording macros, should I continue reporting or wait for this one to be fixed ?

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: ProDark.qss/ProDark/Fusion
Logical/physical DPI: 96/93.338
Installed mods: 
  * Assembly3 0.12.2
  * CurvesWB 0.6.12
  * Defeaturing 1.2.3
  * ExtremeProDark 2.6.4
  * FreeCAD-Ribbon 1.6.2.6
  * FreeCAD-themes 2024.7.24
  * freecad-xr-workbench 0.9.4
  * freecad_streamdeck_addon 0.1.7
  * OpenDark 2024.3.13
  * OpenTheme 2024.9.1
  * parts_library
  * parts_library.bak
  * ProDarkThemePreferencePack 1.0.0
  * SearchBar 1.3.3
  * workfeature
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-12:
Is there an issue in the code? I don't see datum objects in the macro or file. Could you ask on the forum about that?

mgth, 2025-02-12:
I had no error while creating the macro, but at replay.
I don't need help with this. I just wanted to point out a case where macro won't playback. (And that's why I asked if we should continu reporting on Macro cause I had many other cases like this one.)

At line 95 :
```
App.getDocument('Unnamed').getObject('Sketch').setDatum(10,App.Units.Quantity('20.000000 mm'))
```
Crashes with 

11:19:06  ValueError: Invalid constraint index: 10

If you look at the two lines before : 

```
App.getDocument('Unnamed').getObject('Sketch').addConstraint(Sketcher.Constraint('Distance',0,1,0,2,20.000000)) 
App.getDocument('Unnamed').getObject('Sketch').addConstraint(Sketcher.Constraint('DistanceY',0,2,0,1,20.000000)) 

```

constraints are created that seams to be the proposed default rectangle sizes witch I did not validate at creation.
If I remove them, I still get an error (maybe because the index has changed).

FlachyJoe, 2025-02-12:
A recording macro save the process while it executes. It exists obvious cases where the process can't be done anymore:

- Create an object
- Start to record a macro
- Change the object
- Stop the recording
- Delete the object
- The macro can't execute without error due to the object missing

You shouldn't try to launch a recorded macro without understanding in the code.

closed 2025-02-12 by maxwxyz