FreeCAD/FreeCAD #19446: the  COOLANT doesn't work refactored_grbl_post.py --> Missing values["ENABLE_COOLANT"] = True

PatrikRindlisbacher, 2025-02-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

the  COOLANT doesn't work refactored_grbl_post.py 

--> Missing in Postprozessor  --> **values["ENABLE_COOLANT"] = True**

After a long search I found the error

### Full version info

```shell
OS: Windows 11 build 22621
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

LarryWoestman, 2025-02-07:
It is not a bug if an existing postprocessor file doesn't happen to have the options set the way your particular machine needs them :-)  The grbl post file (refactored or not) is pretty "generic" and works for some machines but won't be correct for others.  Many grbl-based machines don't support using coolant at all. Some do.  One postprocessor file won't support all of the possibilities.

You might consider creating a "custom" postprocessor file that has all of the options set the way that your machine wants.  It is currently easier to do this with the old-style (non-refactored) postprocessor files.  The refactored postprocessors lost much of their customizability when I moved a lot of shared code to the Utils* files, and I am not done with the changes necessary to add a fair amount of customizabilty back.  That is one of the reason that the refactored postprocessors are still "under construction".

PatrikRindlisbacher, 2025-02-07:
I intend to improve the quality of Free CAD with my contributions.

[image]

LarryWoestman, 2025-02-07:
You contributions are welcome :-)  That's how FreeCAD improves, as you said.

I believe what you are reporting with your most recent post is another issue:  using an array doesn't handle coolant correctly?

I would file a separate github issue for that.  Otherwise things get confused :-)

phaseloop, 2025-03-20:
@LarryWoestman 

I would agree with @PatrikRindlisbacher that this is a bug - "refactored" grbl should have same behaviour as "original" grbl postprocessor (which handles coolant properly).

GRBL supports coolant g-code as a part of supported command list. Even if many people do not have coolant pump - many connect there a vacuum or air compressor.

LarryWoestman, 2025-03-26:
PR #20488 enables coolant-related commands in the refactored_grbl postprocessor.

closed 2025-03-27 by hyarion