FreeCAD/FreeCAD #19448: Grossly erroneous mill bit diameter parameter in CAM job results in loss of items from viewport

Bazmundi, 2025-02-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I was setting up a job and changed tool from Endmill to Ballnose and was setting diameter but missed the decimal point and typed in 3176mm. OH NO! 

Net end effect.  Blew out dimentions of mill bit of course. 

When I readjusted the diameter of the bit in the job wizard, the model represented by "body" was no longer in the viewport!!?!?!  

I went to the viewport and CTRL-A and trying to zoom to selected did not work, nothing to zoom to (and as likely nothing to select).  

No amount of toggling of visibility of actual body nor body copy under job seemed to recover the model into the viewport.

I exited without saving then restarted FreeCAD.

Whether or not the following are related, sorry to report:

1. When I now click on the EndMill under TC:DefaultTool it now reports "I kenna find the shapefile".
2. I am also getting the following exception(s):
```
07:50:02  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1668, in clearSelection
    self.updateSelection()
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1438, in updateSelection
    self.form.setOrigin.setEnabled(False)
    ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'setOrigin'
07:50:02  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1659, in addSelection
    self.updateSelection()
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1438, in updateSelection
    self.form.setOrigin.setEnabled(False)
    ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'setOrigin'
07:50:11  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1668, in clearSelection
    self.updateSelection()
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1438, in updateSelection
    self.form.setOrigin.setEnabled(False)
    ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'setOrigin'
07:50:11  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1659, in addSelection
    self.updateSelection()
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1438, in updateSelection
    self.form.setOrigin.setEnabled(False)
    ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'setOrigin'
07:50:16  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1668, in clearSelection
    self.updateSelection()
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1438, in updateSelection
    self.form.setOrigin.setEnabled(False)
    ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'setOrigin'
07:50:16  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1659, in addSelection
    self.updateSelection()
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1438, in updateSelection
    self.form.setOrigin.setEnabled(False)
    ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'setOrigin'
07:59:39  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1668, in clearSelection
    self.updateSelection()
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1438, in updateSelection
    self.form.setOrigin.setEnabled(False)
    ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'setOrigin'
07:59:39  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1659, in addSelection
    self.updateSelection()
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1438, in updateSelection
    self.form.setOrigin.setEnabled(False)
    ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'setOrigin'
07:59:41  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1668, in clearSelection
    self.updateSelection()
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1438, in updateSelection
    self.form.setOrigin.setEnabled(False)
    ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'setOrigin'
07:59:41  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1659, in addSelection
    self.updateSelection()
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1438, in updateSelection
    self.form.setOrigin.setEnabled(False)
    ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'setOrigin'
07:59:44  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1668, in clearSelection
    self.updateSelection()
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1438, in updateSelection
    self.form.setOrigin.setEnabled(False)
    ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'setOrigin'
07:59:44  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1659, in addSelection
    self.updateSelection()
  File "/tmp/.mount_FreeCACeiOtg/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1438, in updateSelection
    self.form.setOrigin.setEnabled(False)
    ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'setOrigin'

```

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Manipulator 1.5.7
  * Design456 0.0.1
  * FeedsAndSpeeds 0.6.0
  * LCInterlocking
  * Assembly4 0.50.16-dev
  * lattice2 1.0.0
  * fasteners 0.5.37
  * PieMenu 1.2.7
  * Curves 0.6.55
  * ThreadProfile 1.92.0
  * parts_library
  * freecad.gears 1.3.0
  * A2plus.backup1728897810.7832332 0.4.61 (Disabled)
  * Nodes 0.1.36
  * kicadStepUpMod 11.5.2
  * Beltrami 1.1.0
  * ImportNURBS 1.1.0
  * sheetmetal 0.7.5
  * A2plus 0.4.68
  * Gridfinity 0.8.3
```

### Subproject(s) affected?

CAM

### Anything else?

This is the first job I am setting up in FreeCAD 1.0.0RC2 and it's breaking in the first few minutes etc.  I am assuming something under the hood is now broken if the Endmill shapefile are missing now or at least the paths to files have somehow been broken?

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-31:
@Bazmundi can you provide example files for testers to test ? Any chance you could also test on latest 1.1-dev ?

tarman3, 2025-05-31:
There is a new tool library will be soon
So new bugs will replace old bugs ))