FreeCAD/FreeCAD #19449: ControlSingleton::showDialog: Can't show  Gui::TaskView::TaskDialog  since there is already an active task dialog

leoheck, 2025-02-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Freecad is crashing a lot.
I don't know how to reproduce this... I was about to edit the pad height, using formula/equation directly in the Properties view.

```
ControlSingleton::showDialog: Can't show  Gui::TaskView::TaskDialog  since there is already an active task dialog
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x70a4f5e45320]
#1  0x70a4f74fa9a6 in QDialogButtonBox::button(QDialogButtonBox::StandardButton) const from /tmp/.mount_FreeCAPimhNC/usr/bin/../lib/libQt5Widgets.so.5+0x6
#2  0x70a4c60c4bef in Gui::TaskMeasure::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCAPimhNC/usr/lib/MeasureGui.so+0xbf
#3  0x70a4f64b67ca in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAPimhNC/usr/bin/../lib/libQt5Core.so.5+0x6a
#4  0x70a4f73703e8 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAPimhNC/usr/bin/../lib/libQt5Widgets.so.5+0xb8
#5  0x70a4f9174f79 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAPimhNC/usr/bin/../lib/libFreeCADGui.so+0xc9
#6  0x70a4f64b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAPimhNC/usr/bin/../lib/libQt5Core.so.5+0x108
#7  0x70a4f6b312fb in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /tmp/.mount_FreeCAPimhNC/usr/bin/../lib/libQt5Gui.so.5+0xdb
#8  0x70a4f6b0b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAPimhNC/usr/bin/../lib/libQt5Gui.so.5+0xac
#9  /tmp/.mount_FreeCAPimhNC/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x70a4f0b13bea]
#10  /tmp/.mount_FreeCAPimhNC/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x70a4f4710ad7]
#11  /tmp/.mount_FreeCAPimhNC/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x70a4f4713ac7]
#12  /tmp/.mount_FreeCAPimhNC/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x70a4f47140ce]
#13  0x70a4f650aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAPimhNC/usr/bin/../lib/libQt5Core.so.5+0x66
#14  0x70a4f64b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAPimhNC/usr/bin/../lib/libQt5Core.so.5+0x11b
#15  0x70a4f64bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAPimhNC/usr/bin/../lib/libQt5Core.so.5+0x82
#16  0x70a4f90d4f7a in Gui::Application::runApplication() from /tmp/.mount_FreeCAPimhNC/usr/bin/../lib/libFreeCADGui.so+0x8aa
#17  /tmp/.mount_FreeCAPimhNC/usr/bin/freecad(+0x78f6) [0x5ac31670c8f6]
#18  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x70a4f5e2a1ca]
#19  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x70a4f5e2a28b]
#20  /tmp/.mount_FreeCAPimhNC/usr/bin/freecad(+0x7c7e) [0x5ac31670cc7e]
```

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.7.5
  * Curves 0.6.54
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-07:
Please add steps how to reproduce.
related: https://github.com/FreeCAD/FreeCAD/issues/11914

maxwxyz, 2025-02-08:
related https://github.com/FreeCAD/FreeCAD/issues/18359 ?

leoheck, 2025-09-26:
I haven't observed this anymore.

The related issues are also addressing my need, which is to make tabs independent editors. This should allow multiple task editors to be opened at the same time instead of having to close one to work on another file, which is a pain in the ass for the design flow. This should also support having multiple and independent report views, one for each open file.