FreeCAD/FreeCAD #19451: Problem when entering Draft line length using the property view

towheed, 2025-02-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If the same length is entered in the property view panel as the existing length, the length changes to a different value.

Steps to re-create:

1.  Switch to Draft/BIM workbench
2. Draw a line of any random length
3. Triple-click on the length property in the property view thus selecting both the value and units.  Change it to say 100mm.  Assuming the original length was not 100mm, in my case it changed to 100.00mm.  Normal behavior.
4. Triple-click again on the length property and enter 100 again.  It changes to 10.00mm.  Rinse and repeat and the length alternates between 100 and 10.

This does not happen with the Rectangle or Circle tools.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.36
  * sheetmetal 0.7.5
  * frame 0.1.1
  * parts_library
  * Reinforcement
  * symbols_library
```

### Subproject(s) affected?

None

### Anything else?

I've noticed the same behavior on 0.21.2.

OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * kicadStepUpMod 11.2.4
  * symbols_library
  * dodo 1.0.1
  * BIM 2021.12.0
  * 3DfindIT 1.2.0
  * freecad.gears 1.2.0
  * fasteners 0.5.24
  * pcb 6.2023.1

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-02-07:
You have to enter not only the value but also the unit, in your case you have to write 100 mm, if you write only 100 it produces 10mm.  In fact, FC can't remember the defult unit system. If the default is cm and if I enter 100 it means 100 cm but now we have to enter 100 cm(number+unit) as far as I know.I think this should be a bug

maxwxyz, 2025-02-07:
@Roy-043 FYI

PR #19461 (2025-02-07): Draft: Fix line length input issue
Fixes #19451.
commit 4f45d9b2 (2025-02-07): Draft: Fix line length input issue (#19461)

Fixes #19451.
commit 8b124b65 (2025-02-07): Draft: Fix line length input issue

Fixes #19451.
commit 79655aab (2025-02-16): Draft: Fix line length input issue

Fixes #19451.
closed 2025-02-07 by Roy-043