FreeCAD/FreeCAD #19452: Segfault when removing an object attached to a LCS.

leoheck, 2025-02-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Segfault when removing an object attached to a  LCS.
These little things are infuriating.

```
'NoneType' object has no attribute 'ViewObject'Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x752445845330]
#1  0x7523fe3f7882 in TopLoc_Location::Multiplied(TopLoc_Location const&) const from /tmp/.mount_FreeCAMJKDGm/usr/lib/./libTKMath.so.7.7+0x12
#2  0x752421351ec0 in BRep_Tool::Surface(TopoDS_Face const&, TopLoc_Location&) from /tmp/.mount_FreeCAMJKDGm/usr/lib/./libTKBRep.so.7.7+0x30
#3  0x752421366a17 in BRepAdaptor_Surface::Initialize(TopoDS_Face const&, bool) from /tmp/.mount_FreeCAMJKDGm/usr/lib/./libTKBRep.so.7.7+0x97
#4  0x752421367d23 in BRepAdaptor_Surface::BRepAdaptor_Surface(TopoDS_Face const&, bool) from /tmp/.mount_FreeCAMJKDGm/usr/lib/./libTKBRep.so.7.7+0xf3
#5  0x7524230ca809 in Measure::Measurement::planePlaneDistance() const from /tmp/.mount_FreeCAMJKDGm/usr/lib/Measure.so+0xc9
#6  0x75240bead57e in MeasureGui::QuickMeasure::printResult() from /tmp/.mount_FreeCAMJKDGm/usr/lib/MeasureGui.so+0x42e
#7  0x75240beade0d in MeasureGui::QuickMeasure::processSelection() from /tmp/.mount_FreeCAMJKDGm/usr/lib/MeasureGui.so+0x5d
#8  /tmp/.mount_FreeCAMJKDGm/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x752445eeb6f2]
#9  0x752445eef34a in QTimer::timeout(QTimer::QPrivateSignal) from /tmp/.mount_FreeCAMJKDGm/usr/bin/../lib/libQt5Core.so.5+0x2a
#10  0x752445ee064d in QObject::event(QEvent*) from /tmp/.mount_FreeCAMJKDGm/usr/bin/../lib/libQt5Core.so.5+0x7d
#11  0x752446d703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAMJKDGm/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#12  0x752448b74f79 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAMJKDGm/usr/bin/../lib/libFreeCADGui.so+0xc9
#13  0x752445eb6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAMJKDGm/usr/bin/../lib/libQt5Core.so.5+0x108
#14  0x752445f09f51 in QTimerInfoList::activateTimers() from /tmp/.mount_FreeCAMJKDGm/usr/bin/../lib/libQt5Core.so.5+0x381
#15  /tmp/.mount_FreeCAMJKDGm/usr/bin/../lib/libQt5Core.so.5(+0x30a75c) [0x752445f0a75c]
#16  /tmp/.mount_FreeCAMJKDGm/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x752444110ad7]
#17  /tmp/.mount_FreeCAMJKDGm/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x752444113ac7]
#18  /tmp/.mount_FreeCAMJKDGm/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7524441140ce]
#19  0x752445f0aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAMJKDGm/usr/bin/../lib/libQt5Core.so.5+0x66
#20  0x752445eb570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAMJKDGm/usr/bin/../lib/libQt5Core.so.5+0x11b
#21  0x752445ebd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAMJKDGm/usr/bin/../lib/libQt5Core.so.5+0x82
#22  0x752448ad4f7a in Gui::Application::runApplication() from /tmp/.mount_FreeCAMJKDGm/usr/bin/../lib/libFreeCADGui.so+0x8aa
#23  /tmp/.mount_FreeCAMJKDGm/usr/bin/freecad(+0x78f6) [0x652ea49928f6]
#24  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x75244582a1ca]
#25  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x75244582a28b]
#26  /tmp/.mount_FreeCAMJKDGm/usr/bin/freecad(+0x7c7e) [0x652ea4992c7e]
```

This one happened after recovering the files and starting to work with what I've lost. Crazy.
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x7cb92e245330]
#1  0x7cb8e23f7882 in TopLoc_Location::Multiplied(TopLoc_Location const&) const from /tmp/.mount_FreeCAaGCGmA/usr/lib/./libTKMath.so.7.7+0x12
#2  0x7cb908351ec0 in BRep_Tool::Surface(TopoDS_Face const&, TopLoc_Location&) from /tmp/.mount_FreeCAaGCGmA/usr/lib/./libTKBRep.so.7.7+0x30
#3  0x7cb908366a17 in BRepAdaptor_Surface::Initialize(TopoDS_Face const&, bool) from /tmp/.mount_FreeCAaGCGmA/usr/lib/./libTKBRep.so.7.7+0x97
#4  0x7cb908367d23 in BRepAdaptor_Surface::BRepAdaptor_Surface(TopoDS_Face const&, bool) from /tmp/.mount_FreeCAaGCGmA/usr/lib/./libTKBRep.so.7.7+0xf3
#5  0x7cb911aca809 in Measure::Measurement::planePlaneDistance() const from /tmp/.mount_FreeCAaGCGmA/usr/lib/Measure.so+0xc9
#6  0x7cb8fc04a57e in MeasureGui::QuickMeasure::printResult() from /tmp/.mount_FreeCAaGCGmA/usr/lib/MeasureGui.so+0x42e
#7  0x7cb8fc04ae0d in MeasureGui::QuickMeasure::processSelection() from /tmp/.mount_FreeCAaGCGmA/usr/lib/MeasureGui.so+0x5d
#8  /tmp/.mount_FreeCAaGCGmA/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7cb92e8eb6f2]
#9  0x7cb92e8ef34a in QTimer::timeout(QTimer::QPrivateSignal) from /tmp/.mount_FreeCAaGCGmA/usr/bin/../lib/libQt5Core.so.5+0x2a
#10  0x7cb92e8e064d in QObject::event(QEvent*) from /tmp/.mount_FreeCAaGCGmA/usr/bin/../lib/libQt5Core.so.5+0x7d
#11  0x7cb92f7703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAaGCGmA/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#12  0x7cb931574f79 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAaGCGmA/usr/bin/../lib/libFreeCADGui.so+0xc9
#13  0x7cb92e8b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAaGCGmA/usr/bin/../lib/libQt5Core.so.5+0x108
#14  0x7cb92e909f51 in QTimerInfoList::activateTimers() from /tmp/.mount_FreeCAaGCGmA/usr/bin/../lib/libQt5Core.so.5+0x381
#15  /tmp/.mount_FreeCAaGCGmA/usr/bin/../lib/libQt5Core.so.5(+0x30a75c) [0x7cb92e90a75c]
#16  /tmp/.mount_FreeCAaGCGmA/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7cb92cb10ad7]
#17  /tmp/.mount_FreeCAaGCGmA/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7cb92cb13ac7]
#18  /tmp/.mount_FreeCAaGCGmA/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7cb92cb140ce]
#19  0x7cb92e90aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAaGCGmA/usr/bin/../lib/libQt5Core.so.5+0x66
#20  0x7cb92e8b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAaGCGmA/usr/bin/../lib/libQt5Core.so.5+0x11b
#21  0x7cb92e8bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAaGCGmA/usr/bin/../lib/libQt5Core.so.5+0x82
#22  0x7cb9314d4f7a in Gui::Application::runApplication() from /tmp/.mount_FreeCAaGCGmA/usr/bin/../lib/libFreeCADGui.so+0x8aa
#23  /tmp/.mount_FreeCAaGCGmA/usr/bin/freecad(+0x78f6) [0x5e7e5c9c18f6]
#24  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x7cb92e22a1ca]
#25  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7cb92e22a28b]
#26  /tmp/.mount_FreeCAaGCGmA/usr/bin/freecad(+0x7c7e) [0x5e7e5c9c1c7e]
```

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.7.5
  * Curves 0.6.54
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-07:
@PaddleStroke FYI

leoheck, 2025-02-21:
This may be linked.
As soon as I click the LCS object. Freecad crashes.
https://github.com/FreeCAD/FreeCAD/issues/19756

leoheck, 2025-02-21:
I will try to explain the scenario.
I have a Body full of features.
Then I changed the tip to the middle.
After forgetting about this, I added an LCS (2 actually).
So the LCS objects were added to the middle of the Body tree since the tip was not the last object.