FreeCAD/FreeCAD #19455: Techdraw WB does't  recognize the  white color : white line colors are always black in TD

ErtugrulEge, 2025-02-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to re-create:
1.Switch to Draft/BIM workbench
2. Draw a line or rectangle
3. Select your drawing and change the  Object style->line color to white (255,255,255) 
4. Switch to Techdraw workbench and create a page
5. Select your drawing and insert as  draft workbench object(TechDraw_DraftView)  with a suitable scale
6. You will see that the line color is black.

and if you print the document it will be printed as black instead of white

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40006 (Git) Conda
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

TechDraw

### Anything else?

Other colors are working well except white.Sometimes we may want some lines not to appear in the output, so white lines can be useful in some cases

[image]

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-02-08:
In my opinion, default parameter should be false.From a user perspective, if the line color is white, I expect it to be white on the page

Roy-043, 2025-02-08:
Changing the current default can be problematic for existing users who rely on it.

ErtugrulEge, 2025-08-20:
> Changing the current default can be problematic for existing users who rely on it.

Now, I think you are right.Default behaviour is correct also we can adjust the setting. We can close this issue

[image]

closed 2025-08-20 by luzpaz