FreeCAD/FreeCAD #19456: MESH: Import of OBJ result in empty mesh object

rostskadat, 2025-02-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

# Description

I need to import an .OBJ mesh from a file. When open in Blender it shows as a simple cylinder.
However when imported in FC through the Mesh package it seems to fail to import anything and result in an empty mesh.
Unfortunately there are no error messages in the report view.

Here is the mesh as seen in Blender:

[image]

And here is the OBJ file and its associated MTL files

[texturableCylinder.zip](https://github.com/user-attachments/files/18702479/texturableCylinder.zip)

# How to reproduce?

It is really simple to reproduce. In the python console:

```python
>>> import Mesh
>>> Mesh.insert('/path/to/texturableCylinder.obj')
>>> App.ActiveDocument.texturableCylinder.Mesh
Mesh: [0 Faces, 0 Edges, 0 Points]
```

# What was expected?

I was expecting to have the mesh properly load into the `ActiveDocument`

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (xcb)
Architecture: x86_64
Version: 1.1.0dev.39988 (Git)
Build type: Release
Branch: main
Hash: d7f0ce2a97d8338992c45572980cc06274f1c6c6
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Mesh

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-07:
I cannot reproduce by opening it manually as obj

[image]

furgo16, 2025-02-07:
I don't know why FreeCAD does the distinction (both on import and export), as it appears to be the same format:

- https://en.wikipedia.org/wiki/Wavefront_.obj_file
- https://dirsig.cis.rit.edu/docs/new/obj.html

Alias and Wavefront seem to be the two companies that developed the format. If I understand it correctly, the distinction between `Alias Mesh` and `Wavefront OBJ` seems to be simply between the two different modules in FreeCAD that do the import/export

I'm not an expert, but I've been bitten by this a few times. I can understand that there are two importers for legacy reasons, and that it might not be trivial to merge them. By why not call the format simply `.OBJ`?

If a distinction needs to be made, it could still be called `OBJ (Mesh)` and `OBJ (Arch)`.

furgo16, 2025-08-08:
Related: https://github.com/FreeCAD/FreeCAD/issues/21307