FreeCAD/FreeCAD #19459: Draft: polyline preview broken

benj5378, 2025-02-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Expected: seeing the so far drawn polyline.
Actual: no idea what I drew.

https://github.com/user-attachments/assets/77aa86df-2c72-4128-a3b8-331e419a23c6

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40043 (Git) Snap 1293
Build type: Release
Branch: main
Hash: 34d9ac00506ace1b930a4efa075b97bee810f1c7
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/159.34
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

benj5378, 2025-02-07:
I cannot reproduce anymore. It seems like the line having no color (even after adding/closing task), but only under some specific workflow.

maxwxyz, 2025-02-07:
@Roy-043 FYI

Roy-043, 2025-02-07:
Let's close this if it is not reproducible. Such cases should always be tested in safe mode BTW.

closed 2025-02-07 by Roy-043