FreeCAD/FreeCAD #19460: Assembly: Including parts canlead to a crash of the assembly flow when included parts are deleted by hand and not from the Insertion tab

callistopili, 2025-02-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I found an issue on the 1.0 stable, if I create the Assembly and I play with the Insert Component Tab, I can pick the part I need easily, and it is great.

But if I delete the new added part by hand(using the Model View) then the Insert Component table is still keeping the part referred as added,
with the right click of the mouse decrease the quantity only if the part previously inserted still exists, if not, then you are removing something not existing anymore.

I started with this:
[image]

at this point the count is ok, 1 pcs of one part and two pieces of another part
[image]

I deleted by hand the part from the Model View and not by the Component insertion.
You still see the count wrong(which leads to the error next) by trying to remove/add from the Insertion tab it crash the script.
[image]

the script, I see crash because no real reference of the part is existing:
[image]

Hope you can replicate, thx .

Callisto.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * Estimate 0.1.2
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Gridfinity 0.8.0
  * kicadStepUpMod 11.5.2
  * LCInterlocking
  * MakerWorkbench 1.0.1
  * sheetmetal 0.7.5
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-12:
@PaddleStroke FYI

luzpaz, 2025-09-26:
@callistopili can you reproduce on latest dev ? https://github.com/FreeCAD/FreeCAD/releases