FreeCAD/FreeCAD #19462: Core: inconsistent organization of preference pages

benj5378, 2025-02-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The organization of settings of the 3D view are confusing.

- Some colors are set under respective the subpages (3D View), while others under the Color subpage.
- 3D cube preferences are in the Navigation subpage, while the axis system (being very similar to the 3D cube) preferences are located in 3D View subpage.
- The backlight is under 3D View, while the light source is under Light Sources.

There are probably more examples.

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40043 (Git) Snap 1293
Build type: Release
Branch: main
Hash: 34d9ac00506ace1b930a4efa075b97bee810f1c7
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/159.34
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-02-08:
FWIW, TD's preferences are a shambles too.

maxwxyz, 2025-02-10:
Maybe we can have a project to evaluate and write down a proposal for a better logical arrangement @FreeCAD/design-working-group ?

MisterMakerNL, 2025-02-10:
I don't think re-ordering will fix the real issue here.
The real issue here is that there are many gray zones for many settings and it is way to hard to re-order it.
I know programs with a lot of settings that just use a giant table, would it be an idea to simplify stuff and do something like this?

benj5378, 2025-02-10:
> I know programs with a lot of settings that just use a giant table, would it be an idea to simplify stuff and do something like this?

@MisterMakerNL, that has been my proposal on GitHub as well. I think the preference pages should be auto generated, rather than manually designed in .ui files. This might also allow searching. Of course not all preferences can be auto-generated, but those that can, they should.

While there are grayzones, there also inconsistent organization not due to gray zones - as with those mentioned examples: backcolor/lightsources, and navicube/axis cross