FreeCAD/FreeCAD #19465: Geometry issues arises from nothing more frequently than normal.

leoheck, 2025-02-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have issues like this everywhere every time.
It is a bit hard to design something useful... from one day to another things like this appear on existing things that are ok in the day before.
Because of this, I cannot add more features since Freecad complains.

[image]

Right now I just added this base, inside a hollow box, that touches an existing feature.

[image]

Why does this thing happen?
Can I waive it somehow?

[image]

[image]

I never had such issues before with 1.0.
on 0.21, the refine was always False, is it related with the refine thing?

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.7.5
  * Curves 0.6.54
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

None

### Anything else?

Now trying to make a sample of it.
Copied the Body to a new file.
Selected all the features after this problematic one.
Hit delete.
And then SEGFAUL.

```
Lost link to 'XY_Plane180' while loading, maybe an object was not loaded correctly
<Sketch> SketchObject.cpp(9242): External geometry Body#Sketch203.e7 missing reference: Hole007.;Edge9
[... 81 lines cut ...]
#54  /tmp/.mount_FreeCAbCOplE/usr/plugins/platforminputcontexts/libibusplatforminputcontextplugin.so(+0xbf1b) [0x76598fce0f1b]
#55  /tmp/.mount_FreeCAbCOplE/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7659958eb6f2]
#56  0x76598fe8fb5f in QDBusPendingCallWatcher::finished(QDBusPendingCallWatcher*) from /tmp/.mount_FreeCAbCOplE/usr/plugins/platforms/../../lib/libQt5DBus.so.5+0x2f
#57  0x7659958e0830 in QObject::event(QEvent*) from /tmp/.mount_FreeCAbCOplE/usr/bin/../lib/libQt5Core.so.5+0x260
#58  0x7659967703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAbCOplE/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#59  0x765998574f79 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAbCOplE/usr/bin/../lib/libFreeCADGui.so+0xc9
#60  0x7659958b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAbCOplE/usr/bin/../lib/libQt5Core.so.5+0x108
#61  0x7659958b9801 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /tmp/.mount_FreeCAbCOplE/usr/bin/../lib/libQt5Core.so.5+0x181
#62  /tmp/.mount_FreeCAbCOplE/usr/bin/../lib/libQt5Core.so.5(+0x30b313) [0x76599590b313]
#63  /tmp/.mount_FreeCAbCOplE/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x765993b10ad7]
#64  /tmp/.mount_FreeCAbCOplE/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x765993b13ac7]
#65  /tmp/.mount_FreeCAbCOplE/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x765993b140ce]
#66  0x76599590aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAbCOplE/usr/bin/../lib/libQt5Core.so.5+0x66
#67  0x7659958b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAbCOplE/usr/bin/../lib/libQt5Core.so.5+0x11b
#68  0x7659958bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAbCOplE/usr/bin/../lib/libQt5Core.so.5+0x82
#69  0x7659984d4f7a in Gui::Application::runApplication() from /tmp/.mount_FreeCAbCOplE/usr/bin/../lib/libFreeCADGui.so+0x8aa
#70  /tmp/.mount_FreeCAbCOplE/usr/bin/freecad(+0x78f6) [0x5bb45b6c58f6]
#71  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x76599522a1ca]
#72  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x76599522a28b]
#73  /tmp/.mount_FreeCAbCOplE/usr/bin/freecad(+0x7c7e) [0x5bb45b6c5c7e]
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-07:
Please add an example file and steps how to reproduce if this is so simple to reproduce and happens all the time.

leoheck, 2025-09-27:
We can close this, I can reopen it if I needed.