FreeCAD/FreeCAD #19466: BIM_TDPage creates a defective Page without a label when no template is selected

semhustej, 2025-02-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

To reproduce the issue:

- start BIM_TDPage in a new file
- select `Cancel` in the template selection window
- new page is created with no Label
- the template inside of this page has no link to an svg file
- the page is defective - I am not able to drag the view

[image]

**Possible solutions:**

1. When Cancel is selected in template selection window, there should be no page created - the command is canceled. This is how TechDraw_PageTemplate command works and is preferable solution in my point of view
2. When Cancel is selected in template selection window, default page should be created (as with TechDraw_DefaultPage command)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
Installed mods: 
  * symbols_library
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-02-09:
Wouldn't it be better if BIM just used the TD functions?  Might be some c++/py integration to sort out.

semhustej, 2025-02-09:
> Wouldn't it be better if BIM just used the TD functions? Might be some c++/py integration to sort out.

I don't know. I would guess Yorik had a reason to create independent tool. Let's wait for his reaction.

yorikvanhavre, 2025-02-10:
It was just a shortcut to create a page and immediately pop up a file chooser to pick a template, that would remember the last template used. But I added that quite some time ago, and indeed it might be obsolete now.

In any case, it should indeed be fixed so no page is created if no template has been chosen.

closed 2025-02-13 by yorikvanhavre