FreeCAD/FreeCAD #19469: BIM_TDView - if no page is selected, offer user an option on which page to place the view

semhustej, 2025-02-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Feature Request

**Problem:**
When user starts `BIM_TDView` command without selecting a page, the new view is placed on a random (?) page. User has no influence on which page the view is placed.

**Possible solution:**

- if no page is selected, invoke the Page chooser dialogue - the same, that `TechDraw_View` uses in this case. This would add consistency across workbenches

Page chooser dialogue from TechDraw_View

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
Installed mods: 
  * symbols_library
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-02-09:
In TechDraw, we use DlgPageChooser for this.  I don't know if that is accessible from python.

yorikvanhavre, 2025-02-10:
If there is a page in the selection, then the view is added to that page, otherwise it is added to the last page found. But indeed that's not very handy. Indeed we could make the DlgPageChooser available to python, that would be great! I can have a look at it

furgo16, 2025-04-28:
@maxwxyz would you mind reopening this issue? I accidentally linked it to the PR that got merged, but it wasn't actually a fix. Thanks!

maxwxyz, 2026-11-07:
This is implemented in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/db083b4c5243578a7c99ecff7e4d77fcc2ddd6f0

BIM_TDView uses the TechDraw page chooser instead of taking the first page. TechDrawGui.findPage is available from Python too.