FreeCAD/FreeCAD #19474: Toggling visibility of a selected Body by clicking on 3D view is performed on the Feature and not the Body itself.

leoheck, 2025-02-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When toggling visibility of Bodies with spacebar, if the object os selected by clicking on the 3d object itself, the thing that os hidden os the last feature of the Body and not the Body itself. 

Why is that? Any special reason for this behavior?

At the first moment, this is not a problem. But if the object is hidden and the selection changes (for any reason). Then it is quite hard to find the object that is hidden since the Object Tree is showing a Body that is not hidden. And if you have a long list of objects this becomes harder. And you have to unfold and scroll objects to unhide the last feature of them.

Can we have this behavior changed? If the 3D object is selected it should hide the Body itself not the last/active features of it.

### Full version info

```shell
1.0, 1.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-08:
Please check for existing issues

closed 2025-02-08 by maxwxyz

tarman3, 2025-02-08:
I tried to introduce such behavior by macro, which replace **STD_ToogleVisibilty**
I created this for fix behavior with **Job** objects

@leoheck, You can test this

```
"""
Macro to toggle visibility any object, especially Job objects from CAM workbench and its contents
2025-02-08 11:31

https://forum.freecad.org/viewtopic.php?t=94541
https://github.com/FreeCAD/FreeCAD/issues/12453
https://github.com/FreeCAD/FreeCAD/issues/16005

https://wiki.freecad.org/Std_ToggleVisibility
https://wiki.freecad.org/PropertyLink:_InList_and_OutList
https://wiki.freecad.org/Selection_API
"""

hidebaseObject = True   # Hide Object body while hidden Job
showStock = True        # Show Stock while showing the Job

def bodyCheck(body):
    settedTrue = set()
    for obj in body.OutList:
        #print(obj.Name, obj.Visibility)
        if obj.Visibility: settedTrue.add(obj)
    print(settedTrue)
    return settedTrue

def toggleVisibility(obj, forcing=None):
    global recDepth
    if forcing == None:
        obj.Visibility = not obj.Visibility
        output = obj.Visibility
    else:
        obj.Visibility = forcing
        output = forcing
    print(f'{"   "*recDepth} {obj.Name} {recDepth}')
    return output

def recursiveProcessing(o):
    global recDepth
    if o.OutList: recDepth += 1

    for obj in o.OutList:
        if 'Stock' in obj.Name:
            if childState and showStock:
                toggleVisibility(obj, childState)
            else: toggleVisibility(obj, False)

        elif 'SetupSheet' in obj.Name:
            if childState: continue
            else: toggleVisibility(obj, False)

        elif 'Tools' in obj.Name:
            if childState: continue
            else: toggleVisibility(obj, False)

        elif 'ToolController' in obj.Content:
            if childState: continue
            else: toggleVisibility(obj, False)

        elif 'ToolBit' in obj.Name:
            if childState: continue
            else: toggleVisibility(obj, False)

        elif not obj.OutList:
            if not childState and hidebaseObject:
                toggleVisibility(obj, childState)

        else: toggleVisibility(obj, childState)

        if obj.OutList:
            if (not 'Clone' in obj.Label) and (not 'Clone' in obj.Name):
                recursiveProcessing(obj)
            elif hidebaseObject and not childState:
                toggleVisibility(obj.OutList[0], False)

    recDepth -= 1

objSelList = FreeCADGui.Selection.getSelection()

if not objSelList:
    print('Macros VisibilityToggle: nothing selected')

for objSel in objSelList:
    recDepth = 0
    # print(objSel.InList)
    # For Jobs selected
    if 'Job' in objSel.Name:
        childState = toggleVisibility(objSel)
        print(f'Job Toogle visibility from {not childState} to {childState}: {objSel.Name}')
        recursiveProcessing(objSel)

    # For PartDesign::Body
    elif objSel.InList and objSel.InList[0].TypeId == 'PartDesign::Body':
        body = objSel.InList[0]
        # print(body.Name, body.OutList)
        if len(bodyCheck(body)) == 0:
            visibility = toggleVisibility(objSel)
            visibility = toggleVisibility(body, True)
            print(f'Toogle visibility from {not visibility} to {visibility}: {objSel.Name}')

        elif len(bodyCheck(body)) == 1:
            # print('B', bodyCheck(body))
            # print(objSel, bodyCheck(body))
            if objSel in bodyCheck(body):
                visibility = toggleVisibility(body)
                print(f'Toogle visibility from {not visibility} to {visibility}: {body.Name}')
            else:
                visibility = toggleVisibility(objSel)
                print(f'Toogle visibility from {not visibility} to {visibility}: {body.Name}')
        else:
            visibility = toggleVisibility(objSel)
            print(f'Toogle visibility from {not visibility} to {visibility}: {body.Name}')

    # For all other elements
    else:
        visibility = toggleVisibility(objSel)
        print(f'Toogle visibility from {not visibility} to {visibility}: {objSel.Name}')

```

tarman3, 2025-02-08:
https://github.com/user-attachments/assets/d292f7b8-1257-42ef-a6b1-f8c1df805a81

leoheck, 2025-02-13:
@tarman3 thanks. this solution could be added to the core, not as a macro.

However, does it work when you have a link to the Body? And with the original Body hidden and the link visible. If you click the on the 3d model, it should select the last feature of this link, and then the macro should hide the Link. And what if there are 2 links to the body? Does your macro work finding the right Link where the clicked feature belongs?

tarman3, 2025-02-13:
Please attach FCStd file with some examples

leoheck, 2025-02-14:
This is really good @tarman3, good job.