FreeCAD/FreeCAD #19476: TechDraw Arrow Style overwrites existing dimension objects

semhustej, 2025-02-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce

- open FreeCAD
- go to Preferences -> TechDraw -> Dimensions -> change Arrow Style to 'Dot'
- open file (remove the zip file extension)

[FC-TD-dimensions-arrow-style.FCStd.zip](https://github.com/user-attachments/files/18717901/FC-TD-dimensions-arrow-style.FCStd.zip)

- use TechDraw_RedrawPage
- the dimension arrow styles is dots
- close the file
- go to Preferences -> TechDraw -> Dimensions -> change Arrow Style to 'Filled Arrow'
- open the file again
- use TechDraw_RedrawPage
- the dimension arrow styles are filled arrows

The document is influenced by the user setup without user doing anything. This is unwanted behavior I think.

I could reproduce the issue with FreeCAD 1.0 in Safe mode.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
Installed mods: 
  * symbols_library
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

benj5378, 2025-02-08:
Solution to this: on Arrow object initilization, the arrow type should be set according to preferences. Currently, it seem like if no Arrow style is set, the preference is checked on every render.

maxwxyz, 2025-02-08:
@WandererFan is this the intended behavior or a bug?

WandererFan, 2025-02-09:
need ArrowStyle property in ViewProviderDimension
VPD.ArrrowStyle = PreferencesGui::arrowStyle

[QGIViewDimension::drawArrows
-        arrow->setStyle(forcePoint ? 3 : QGIArrow::getPrefArrowStyle());
+       arrow->setStyle(forcePoint ? ArrowType::DOT : vp->ArrowStyle.getValue());

-        if (QGIArrow::getPrefArrowStyle() != 7) {// if not "None"
+        if (vp->ArrowStyle.getValue() != ArrowType::NONE) {// if not "None"](url)

closed 2025-02-23 by WandererFan