FreeCAD/FreeCAD #19482: Assembly Workbench loading all documents when an assembly is opened.

jag45, 2025-02-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Can't get to the forum at the moment so posted here.

See screenshot.
Happens in both 40077 and 40122 - does not happen in previous builds (using the same config files). 
When you open an assembly file containing sub assemblies it opens all of the sub assembly files as well as the main document. This then causes a problem when using 'close all' from the file menu. 
Has there been a setting added which I am not aware of?

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: MY-Darker-Flat.qss/MY-Darker-Flat/Qt default
Logical/physical DPI: 96/102.299
Installed mods: 
  * A2plus 0.4.68
  * Assembly4 0.50.15
  * FreeCAD-themes 2024.7.24
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pierreporte, 2025-02-08:
This is one of the problems I encountered during the holidays.

Ping @PaddleStroke

pierreporte, 2025-02-08:
Related to #18673?

maxwxyz, 2025-02-10:
@PaddleStroke FYI

PaddleStroke, 2025-02-11:
I just came upon this issue after rebasing and it is indeed very annoying. I don't know what has caused this.
I don't think it's related to Assembly workbench. I think it's the partial load that got broken.

PaddleStroke, 2025-03-07:
https://github.com/FreeCAD/FreeCAD/commit/123569823e8c35c698d38ca06a21baaf5a001412 this commit looks very closely related and it in the correct time frame

PaddleStroke, 2025-03-07:
And apparantly it's commit 40058. So right before 40077.

PaddleStroke, 2025-03-07:
@tritao @chennes FYI

PaddleStroke, 2025-03-07:
Great new thanks

PaddleStroke, 2025-03-07:
By the way @tritao , since you have worked with the partial load feature, perhaps you could consider fixing https://github.com/FreeCAD/FreeCAD/issues/18673 as well?
The idea of this issue is that closing a document should also close all the documents that have been partially-opened by this document.

closed 2025-03-08 by chennes