FreeCAD/FreeCAD #19483: Sketcher: Importing external geometry now super error-prone

theosib, 2025-02-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I do like how Sketcher is now able to import external geometry as real objects instead of only construction. But I think the UI could do with some refinement, because I now screw this up literally every time I import external geometry. This is a usability problem.

It's tedious to keep switching to construction geometry and back every time I want to import external geometry as construction, and easy to forget to do it. If I accidentally make new construction geometry when I want real geometry, I can SEE that in blue and fix it right away. On the other hand, if I accidentally import external geometry as non-construction that error is basically invisible because it is coincident with what I just imported. The result is no end of extrusion failures because of stray unclosed geometry that I had wanted for reference.

I have two suggestions on how to fix this:
- Give geometry import its own separate construction mode switch.
- Even better, provide two separate import buttons, one for construction, the other for real geometry.

I also suggest that local geometry (imported or otherwise) should be made more visually distinct compared to geometry of external objects that has not been imported.

### Full version info

```shell
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * freecad.gears 1.3.0
  * Pyramids-and-Polyhedrons
  * fasteners 0.5.29
  * Curves 0.6.50
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

theosib, 2025-02-09:
It's actually really useful to sometimes add external geometry as real geometry. It's just that the way the UI works now, I keep doing it by accident.

maxwxyz, 2025-02-09:
What is the criteria to close this issue? I see no error as described, but rather a sub-optimal UI and users who accidentally clicking the wrong commands.