FreeCAD/FreeCAD #19484: AdditiveHelix causes "Removing splitter failed" at specific turn values

Condyloma, 2025-02-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I’ve encountered an issue with the AdditiveHelix feature in the Part Design workbench. When the helix has less than one full turn (e.g., 0.7 turns), everything works as expected, and additional features like the hole tool function normally. Similarly, if the helix has 3 or more turns, no issues occur.

However, when the helix is set to 1.2 turns, any further modifications, such as adding a hole, result in the following error:

    Removing splitter failed

Strangely, if I increase the turns beyond the height of the object, the issue disappears, and additional features work again.

I’ve also tried modifying the helix profile to ensure that it remains fully inside the solid, but this does not seem to have any effect. The tool seems to be very picky about certain conditions.

Steps to Reproduce

    1. Create a Part Design body.
    2. Use AdditiveHelix with a profile and set it to 1.2 turns.
    3. Try to add a hole or another feature.
    4. Observe the error: Removing splitter failed.
    5. Increase the number of turns beyond the object's height and see that the issue no longer occurs.

Expected Behavior

The model should remain editable regardless of the number of turns, as long as the geometry is valid.

### Full version info

```shell
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.55
```

### Subproject(s) affected?

PartDesign

### Anything else?

Screenshots showing the issue (0.7 and 3 turns work, 1.2 turns fails).
A short video demonstrating the problem.
The FreeCAD file with the problematic model.

0.7 
[image]

3 turn
[image]

1.2 turn
[image]

Link to video:
https://youtu.be/8gvzouirir8

Part:

https://gofile.io/d/sgepNZ

The FreeCAD forum appears to be down, so I’m reporting this here. Is this a known bug, or could it be related to how splitting is handled internally? Any insights would be appreciated.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-02-08:
"Remove splitter failed" is a refine error, please disable the auto-refinement setting.
#17846 #18110

Condyloma, 2025-02-08:
Thank you for the suggestion! Disabling the auto-refinement setting indeed solved the issue. After turning off "Automatically refine model after skitc-based operations", I was able to modify the model without encountering the "Removing splitter failed" error.

I appreciate your help!

closed 2025-02-08 by Condyloma