FreeCAD/FreeCAD #19485: Fillet between 2 arcs in Sketcher

KrisCouvreur, 2025-02-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Fillet between 2 arcs in Sketcher workbench : can someone write the Python code for this ?
In attachment the math behind this.
Pull request was not allowed.

[Fillet between 2 arcs.pdf](https://github.com/user-attachments/files/18720781/Fillet.between.2.arcs.pdf)

### Full version info

```shell
Being able to create ab fillet between 2 arcs would make Sketcher more userfriendly.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-02-13:
This is a good idea for a GSoC project, I'm adding it the GSoC tag

pitch-circle, 2025-02-14:
Looking at the end result in your attached pdf, this is possible using the existing polyline tool with the 'M' modifier.

You can create the three arcs directly from a blank sheet.

If you have two of them already made, and wish to add a fillet later on, turn them into construction geometry and trace over them.

You can also do this with the standard arc tool, applying tangent constraints at the endpoints of the fillet arc. Encompassing the actions needed to do this manually in a macro (i,e, a geometry solution) means that none of the mathematics (an algebra solution) in the pdf is required to be programmed.

pitch-circle, 2025-02-14:
Attached some Python that does as you request (change file extension from txt to py). It uses a different method to your document, with points represented as complex numbers, which simplifies the maths.

FreeCAD file also attached with reference dimensions to confirm Python output. Again, change the extension to FCStd. If experimenting, change the values by a small amount, or the Sketch will flip, confusing things considerably.

[arc fillet.zip](https://github.com/user-attachments/files/18803140/arc.fillet.zip)
[arcfillet.txt](https://github.com/user-attachments/files/18803135/arcfillet.txt)