FreeCAD/FreeCAD #19486: loss of main sight during construction

ThierryRouzies, 2025-02-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I was building a piece yeah the enssemble disappeared, zoom + - unenforceable. It would be desirable to have a command that again allows the display of the assembly. I couldn't get the view of the room.

### Full version info

```shell
0.21
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-02-08:
Have you verified of the objects were still visible by checking the Tree view? Have you tried [Std_ViewFitAll](https://wiki.freecad.org/Std_ViewFitAll)?

maxwxyz, 2025-02-09:
Please test on the stable 1.0 and provide full about info

luzpaz, 2025-09-26:
@ThierryRouzies ping