FreeCAD/FreeCAD #19488: changing Light source on Display Preference, cause the "mouse lamp" movement to be unreachable

callistopili, 2025-02-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I found a minor issue on FreeCAD 1.0  where could happen if you want to change the light  source position.
by going  tinot hte Menu Edit -> Preferences -> tab display -> light sources you get the controls for the light, by mouse or by putting the numbers by hand.

[image]

If you move the mouse and by accident is left on the back of the mock-up part you get a bug of the mouse control.
you see the mouse cannot access autonomously and remain stuck.
[image]

IF you click CANCEL or close the windoe you are OK and the mistake can be recovered next time you open the dialog box.
IF click Apply then it will be stored with the position stuck.

The workaround at the moment is to change by hand the numbers and you need to move the quantity on one of the axis to restore the light pointer nearby out of the back of the centre view of the mock-up part.
 
[image]

### Full version info

```shell
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Dracula 0.0.5
  * kicadStepUpMod 10.23.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-02-08:
How can the solution to the problem look like? And what is exactly the problem? The spinboxes work as expected.

maxwxyz, 2025-02-09:
You can also grab the dragger at the little tip to the opposite of the arrow, so this is always possible.

closed 2025-02-09 by maxwxyz