FreeCAD/FreeCAD #19490: Workaround for 15130 broken: Can't use Cmd-Backquote to switch to hidden window

theosib, 2025-02-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

My bug https://github.com/FreeCAD/FreeCAD/issues/15130 is still open, but the 1.1.x weeklies have made this much worse in a way that I think this deserves a bug report of its own.

The short version if 15130 is that when a window pops up to allow selection of attachment support, clicking in the 3D view causes the selection window to get put behind the main window in the stacking order.

The workaround has always been to press Command-Backquote (the key to the left of the 1 key). On a Mac, this cycles the window stacking order for the active application.

The latest builds have made this much worse by somehow disabling Command-Backquote. So now I have to drag the main window around in order to get at the pop-up.

### Full version info

```shell
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * freecad.gears 1.3.0
  * Pyramids-and-Polyhedrons
  * fasteners 0.5.29
  * Curves 0.6.50
```

### Subproject(s) affected?

Part

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-09:
@JohnOFCII can you reproduce this?

JohnOCFII, 2025-02-09:
> [@JohnOCFII](https://github.com/JohnOCFII) can you reproduce this?

Command-backquote worked for me to get the window to come forward.  I agree with the initial bug report that it should not be drawn behind the main window.

```
OS: macOS 15.3
Architecture: arm64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 72/127.5
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
```

[image]

theosib, 2025-02-09:
This was a persistent problem in 39896. But the problem appears to be resolved in 40122. Please accept my apologies for being lazy about updating before filing the bug report.

closed 2025-02-09 by theosib