FreeCAD/FreeCAD #19493: TechDraw should be buildable without Spreadsheet

benj5378, 2025-02-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

TechDraw includes functionality that links to Spreadsheet, Import and Measure. However for building, TechDraw should be buildable without and not require Spreadsheet, Import or Measure.

TechDraw requires Spreadsheet, Import and Measure. Spreadsheet required Draft.

Suddenly there are no independent build targets and one has to compile all of FreeCAD just to test or develop a specific workbench.

### Full version info

```shell
20ac96f4c8a61eb5c8f3c1f78faa8c88a569165c
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-02-09:
How does ViewSpreadsheet or the export functions work then?

TechDraw uses Measure for 3d Dimensions.

benj5378, 2025-02-09:
@WandererFan they won't. So they should be disabled if the target is not build