FreeCAD/FreeCAD #19494: IFC-Export creates wrong oriented faces

mozman, 2025-02-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I created a simple body in the PartDesign workbench:

[image]

and exported it as IFC file.

This is the result in various CAD applications and IFC viewers:

BricsCAD:

[image]

CAD Assistant:

[image]

Bonsai in Blender with activated face orientation, which shows the problem. There is a complex face with wrong orientation.

[image]

## Possible Solution

I've tried many, many, many fixes to solve this problem, but nothing has worked. My last resort was to remove all existing orientation detections in the `getRepresentation()` function in `exportIFC.py`, and this has worked so far:

[image]

The edited code lines 2280 - 2302 for FreeCAD v1.0 (a patch file is attached):

```Python
for fcface in fcsolid.Faces:
	loops = []
	verts = [v.Point for v in fcface.OuterWire.OrderedVertexes]
	if len(verts) < 1:
		print("Warning: OuterWire returned no ordered Vertexes in ", obj.Label)
		continue
	pts =   [ifcbin.createIfcCartesianPoint(tuple(v)) for v in verts]
	loop =  ifcbin.createIfcPolyLoop(pts)
	bound = ifcfile.createIfcFaceOuterBound(loop,True)
	loops.append(bound)
	for wire in fcface.Wires:
		if wire.hashCode() == fcface.OuterWire.hashCode():
			continue
		verts = [v.Point for v in wire.OrderedVertexes]
		if len(verts) > 1:
			pts =   [ifcbin.createIfcCartesianPoint(tuple(v)) for v in verts]
			loop =  ifcbin.createIfcPolyLoop(pts)
			bound = ifcfile.createIfcFaceBound(loop,True)
			loops.append(bound)
		else:
			print("Warning: wire with one/no vertex in ", obj.Label)
	face =  ifcfile.createIfcFace(loops)
	faces.append(face)
```

Further testing may be necessary.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Austria (de_AT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.68
  * Assembly4 0.50.16-dev
  * fasteners 0.5.37
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Anything else?

`Raster.zip` contains the `Raster.FCStd` file for testing.

[Raster.zip](https://github.com/user-attachments/files/18724407/Raster.zip)

[exportIFC.patch](https://github.com/user-attachments/files/18724384/exportIFC.patch)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

mozman, 2025-02-10:
Addendum:

The new `IfcOpenShell` serializer does it better:

[image]

but has orientation problems with arcs:

FreeCAD:

[image]

Blender-Bonsai:

[image]

BricsCAD:

[image]

[Cross.zip](https://github.com/user-attachments/files/18731470/Cross.zip)

yorikvanhavre, 2025-02-10:
Indeed the serializer has issues with curves, that's a known problem in IfcOpenShell.

Your patch needs indeed to be tested thoroughly, because the current vertex order reversion has been there for a reason too. The thing is, up to your example it had worked flawlessly, I haven't seen a face inversion problem in a very long time...

mozman, 2025-02-10:
@yorikvanhavre I had more problems, especially complex faces (concave faces) are sometimes inverted. The cross product method to detect the face normal does not work reliable for concave faces.  

[image]

[SFH213_9C.zip](https://github.com/user-attachments/files/18731970/SFH213_9C.zip)

I used only the DAE triangulation, may be the internal tesselation works different and returns flipped faces where the DAE triangulation returns correct oriented faces.

luzpaz, 2025-09-26:
@mozman does this still happen on latest 1.1-dev ? https://github.com/FreeCAD/FreeCAD/releases

mozman, 2025-09-26:
@luzpaz In the latest build 43071 the issue is still present:

[image]

and the patch still works:

[image]