FreeCAD/FreeCAD #19495: Draft Shape from Text improvements

leoheck, 2025-02-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

`Draft > Shape From Text` does not accept VarSet.
Having the VarSet here could improve the potential of this tool.
It could be in any of those files.

Coordinates, Text, Height, and Font 

[image]

Extra Note: Why the heck Y is always in `Km` there? This is horrible.

Also, it would be nice to be able to change the reference point. 
It could be: Bottom Left (as is now), Mid Left, Top Left, Bottom Mid, Center, Top Mid, Bottom Right, Mid Right, or Top Right. MetalSheet (Add base shape) has an interesting widget for this, see:

[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.16
  * sheetmetal 0.7.5
  * Curves 0.6.54
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-02-09:
[image]

leoheck, 2025-02-09:
Interesting, but why does the widget that creates it does not support it?

pitch-circle, 2025-02-13:
> ....why does the widget that creates it does not support it?

Many, many 'widgets' throughout FC are not accessible to Expressions. Hence, the challenge you have is not specific to ShapeString.

As the reply above shows, once the item is created, all its Properties are open to Expressions in the Data panel.

The Y-value error could be a floating point under- or over-flow error. See issue 17879 for something very similar.

Roy-043, 2025-02-23:
There is this development: #16928.