FreeCAD/FreeCAD #19496: TechDraw Insert SVG Symbol default folder

semhustej, 2025-02-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature Request**

When invoking TechDraw -> TechDraw Views -> Insert SVG Symbol , user is by default placed in the folder data\Mod\TechDraw\Templates to select the symbol. The same happens when the command is invoked through `TechDraw_View` tool with no objects selected.

**Possible solution:**
The default folder should be set to data\Mod\TechDraw\Symbols as this is much more appropriate for the tool.

I understand, that there is a setting in the TD workbench options to set the directory, but it would be helpful if that setting is empty to be placed in the default symbols directory.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
Installed mods: 
  * symbols_library
```

### Subproject(s) affected?

TechDraw

### Anything else?

I think TechDraw -> TechDraw Views -> Insert Birmap Image should point by default to the same folder.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-02-09:
QString Preferences::defaultSymbolDir()
{
-    std::string defaultDir = App::Application::getResourceDir() + "Mod/TechDraw/Templates";
+    std::string defaultDir = App::Application::getResourceDir() + "Mod/TechDraw/Symbols";

WandererFan, 2025-02-10:
Beat me!  Thanks!

PR #19507 (2025-02-14): TechDraw: Set symbol dir
Fixes #19496
commit 4473aa02 (2025-02-14): TechDraw: Set symbol dir

Fixes #19496
closed 2025-02-14 by chennes