FreeCAD/FreeCAD #19497: Sketcher: Toggle driving/reference constraint does not indicate which mode you're in

theosib, 2025-02-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The button in sketcher to "toggle driving/reference constraint" does not visually indicate whether the current mode is driving or reference. The only way to tell is to try something.

### Full version info

```shell
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Logical/physical DPI: 72/127.5
Installed mods: 
  * freecad.gears 1.3.0
  * Pyramids-and-Polyhedrons
  * fasteners 0.5.29
  * Curves 0.6.50
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

theosib, 2025-02-09:
Evidently the icon does change, but I had to put on some strong reading glasses to see it. Besides, there's nothing intuitive about the colors of the long and short errors. What do they mean?

Maybe the solution would be to make the buttons in the constraints toolbar turn blue.

[image]

maxwxyz, 2025-02-10:
Plus, all other icons which create geometry changes. I'll close this as this is already implemented, even in 1.0

closed 2025-02-10 by maxwxyz