FreeCAD/FreeCAD #19500: CI: Some C++ linter checks are not relevant to this project

chennes, 2025-02-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Right now our CI runs several different linters (cpplint, clang-tidy, and clazy), some of which are using settings that are either in conflict with each other, or that don't reflect this project's coding style. This issue is to collect the suspect settings and get them corrected.

My first pass is that `cpplint` is currently emitting warnings specific to Google's code style that don't apply to us:
 
* **Do not use namespace using-directives.  Use using-declarations instead.  [build/namespaces]** The C++ Core Guidelines permit the use of `using namespace foo` as long as it is local in scope, e.g. confined to *.cpp files. It is a common idiom in our codebase to `use namespace Gui;` for example, in *.cpp files that implement classes in the `Gui` namespace.
* **#ifndef header guard has wrong style, please use: XXX [build/header_guard]** We don't have a standard for what header guards must look like, so this basically just ends up being noise.
* **No copyright message found.  You should have a line: "Copyright [year] <Copyright Owner>"  [legal/copyright]** The scanner here is incorrect, it often fails to detect our normal copyright line so results in a high false positive rate.

`clazy` is currently checking all files, not just changed ones, so its output is too verbose to be useful. `clang-tidy` is in a relatively good state due to its use of the `.clang-tidy.yaml` configuration file at project scope However, the internal CI call for the clang-tidy is quite misleading because it looks like it has its own settings independent of the project's (it used to, but this was eliminated some time ago).

### Full version info

```shell
n/a
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

hyarion, 2025-02-11:
I like the lint check to warn for `using namespace std` :)

It would also be good to stardadize the guards in the code base

chennes, 2025-02-11:
If we can configure one of the linters to catch *just* that one I think it's fine, or to catch any `using namespace` *just* in *.h files, also fine: but `using namespace Gui;` in *.cpp files is too common an idiom in our codebase to consider warning on it.

closed 2025-09-26 by chennes