FreeCAD/FreeCAD #19509: [Part Design] Chamfer appears broken for some shapes

frysztak, 2025-02-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When chamfering three adjacent edges of a cube, all is fine (on the left in the screenshot).
For a more complex shape, chamfer appears to be somewhat broken (on the right in the screenshot)

[image]

I wanted the resulting chamfer to look like this:

[image]

 

### Full version info

```shell
OS: Arch Linux (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.1, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Curves 0.6.53
  * ThreadProfile 1.92.0
  * fasteners 0.5.35
  * A2plus 0.4.68
  * Pyramids-and-Polyhedrons
  * OpenDark 2024.3.13
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Anything else?

Project file: 

[bad_chamfer.zip](https://github.com/user-attachments/files/18731507/bad_chamfer.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-10:
Is it just the appearance or is the shape broken? You can check with Part_CheckGeometry tool.

luzpaz, 2026-09-20:
I can't view the screenshot for some reason, Github is broken atm.

@frysztak can you reproduce this on the latest dev ?

frysztak, 2026-09-24:
@luzpaz tested on weekly-2026.09.23, looks the same as when I reported it (good on the left, bad on the right):

[image]

luzpaz, 2026-10-03:
Is this Coin or OCC related ?

frysztak, 2026-10-03:
@luzpaz are you asking me? I don't know. exported STL also looks bad, so I guess OCC?

luzpaz, 2026-10-03:
cc @pieterhijma