FreeCAD/FreeCAD #19510: PartDesign - DatumLine on cylinder doesn't result as the cylinder axis

ArthurTiV, 2025-02-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When adding a DatumLine on a cylindrical hole pierced on the side (see the pictures below) it will not be the cylinder axis.

the DatumLine moves depending on the position of the hole going through the cylinder. And the angle of the DatumLine depends on that as well.
(Both issues can be seen on the picture below)

Step to reproduce : 
 - Create a sketch with a circle as a hole
 - Extrude
 - create a DatumLine on the cylinder surface
 - create a hole on the side (not on the cylinder's main axis)
 - create a new DatumLine on the cylinder pierced surface
 - you can see that both DatumLines are not the same
 - Move the second hole around
 - You can see the second axis changing

It seems that the DatumLine is some sort of center of mass axis and not getting the axis of the circle from wich the cylinder is created

[image]
[image]
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * AirPlaneDesign 0.4.1
  * CurvedShapes 1.0.13
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-10:
Please share an example file and try it with the new core datums in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

FEA-eng, 2025-11-26:
Still relevant

[Datum line hole center.FCStd.zip](https://github.com/user-attachments/files/23779854/Datum.line.hole.center.FCStd.zip)

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20251126 (Git shallow)
Build date: 2025/11/26 00:30:17
Build type: Release
Branch: (HEAD detached at de230cf)
Hash: de230cfaa75433441366ad09a4d5fda828e7bfdb
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```