FreeCAD/FreeCAD #19511: Spaceball motion does not work in dev version 40122 (3DConnexion)

picli3, 2025-02-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I just installed version 40122 of weekly-builds and the 3D connexion spaceball stopped working, before I was using version 40041 and it works there.

Any ideas on what I can do?

### Full version info

```shell
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Spanish/Peru (es_PE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/86.3799
Installed mods: 
  * sheetmetal 0.7.5
  * ArchTextures
  * fasteners 0.5.37
  * Curves 0.6.54
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-10:
@chennes is this due to the recent PRs?
@pskowronskiTDx FYI

Syres916, 2025-02-10:
@picli3 have you tried ticking the checkbox in `Preferences > Display > Navigation > Space mouse > Enable support of legacy space mouse devices ` ?

picli3, 2025-02-10:
> [@picli3](https://github.com/picli3) have you tried ticking the checkbox in `Preferences > Display > Navigation > Space mouse > Enable support of legacy space mouse devices ` ?

This works, thanks!!

closed 2025-02-10 by maxwxyz

pskowronskiTDx, 2025-02-10:
@maxwxyz As there is no any other solution than "SpNav-legacy" on Linux, I think that newly introduced runtime-switch does not make much sense there.

hkienle, 2025-03-02:
I also had to enable this option:

> > [@picli3](https://github.com/picli3) have you tried ticking the checkbox in `Preferences > Display > Navigation > Space mouse > Enable support of legacy space mouse devices ` ?

I did not know that my 3Dconnexion SpaceMouse Compact is considered legacy! ;)

It would be nice to give the user a hint about this option in case `Customize... > Spaceball *` did not recognize a device!

OS: KDE Flatpak runtime (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.40267 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 04d2bcb)
Hash: 04d2bcbf45ab96f3895ac3d0e8b5ae9becbaf5cd
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical/physical DPI: 96/172.811
Installed mods: 
  * Gridfinity 0.9.7
  * STEMFIE 0.3.1
  * freecad.gears 1.3.0
  * Curves 0.6.58

picli3, 2025-03-02:
I think he is referring to the free driver that they have made to work on Linux.

Because on Windows the driver is quite up to date.

maurerle, 2025-04-27:
I did not have this option in my preferences view:

[image]

The 3Dconnexion compact works fine out of the box on debian trixie with the following packages:

```
freecad/testing,testing,testing,testing,now 1.0.0+dfsg-6 all [installed]
spacenavd/testing,testing,now 1.2-3 amd64 [installed]
```

I do have the settings in Tools -> Customize -> Spaceball motion
So everything as expected, I guess.