FreeCAD/FreeCAD #19514: When measuring things I cannot hide/unhide objects.

leoheck, 2025-02-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When measuring things I cannot hide/unhide objects.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.16
  * sheetmetal 0.7.5
  * Curves 0.6.54
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

Measurement

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-10:
Can you add more context please

leoheck, 2025-02-10:
When I use the Measure tool, I cannot hide/unhide objects.

hyarion, 2025-02-11:
I followed the instructions and it works for me on latest dev build.

Could you describe how you are trying hide/show objects with meassure active?

maxwxyz, 2025-02-11:
Do you have the eye icon turned off in the tree view? Do you try to hide it with space bar?

luzpaz, 2025-05-20:
@leoheck 👆