FreeCAD/FreeCAD #19519: Gesture Navigation causes Add point commmand to error via  Draft_edit

ErtugrulEge, 2025-02-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings to the Community!

if your navigation style is Gesture and if you use draft_edit and inserting  points to a wire (Draft_wire)  that will cause error.

 Steps to re-create:
1.Open a new Draft WB file
2.Be sure your navigation style is Gesture
3.Draw a simple wire (Draft_Wire)
4.Select wire and press edit tool (Draft_Edit)
5.Press Alt+Left click (windows)  opens a menu and selecting Add point causes error

[image]

```
20:09:34  Traceback (most recent call last):
20:09:34    File "C:\Users\CCC\Desktop\FreeCAD_weekly-builds-40122-conda-Windows-x86_64-py311\Mod\Draft\draftguitools\gui_edit.py", line 709, in evaluate_menu_action
20:09:34      callback()
20:09:34    File "C:\Users\CCC\Desktop\FreeCAD_weekly-builds-40122-conda-Windows-x86_64-py311\Mod\Draft\draftguitools\gui_edit.py", line 694, in wrapper
20:09:34      callback()
20:09:34    File "C:\Users\CCC\Desktop\FreeCAD_weekly-builds-40122-conda-Windows-x86_64-py311\Mod\Draft\draftguitools\gui_edit_draft_objects.py", line 115, in <lambda>
20:09:34      (translate("draft", "Add point"), lambda: self.add_point(edit_command, obj, position)),
20:09:34                                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
20:09:34    File "C:\Users\CCC\Desktop\FreeCAD_weekly-builds-40122-conda-Windows-x86_64-py311\Mod\Draft\draftguitools\gui_edit_draft_objects.py", line 147, in add_point
20:09:34      info, newPoint = edit_command.get_specific_object_info(obj,pos)
20:09:34      ^^^^^^^^^^^^^^
20:09:34  TypeError: cannot unpack non-iterable NoneType object
```

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell /.0.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/127.628
```

### Subproject(s) affected?

Draft

### Anything else?

In the menu three commands: Add point,Close wire and Reverse wire. Close and recerse wire commands work but Add point command doesn't work

Forum topic: https://forum.freecad.org/viewtopic.php?t=94665

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-10:
@Rexbas FYI

Roy-043, 2025-02-10:
The problem is related to how a `pivy.coin.SbVec2s` vector is passed. The `pos` vector is not an independent copy. In the case of Gesture navigation this vector gets updated when the mouse is over the menu.
https://github.com/FreeCAD/FreeCAD/blob/516595fbcebaf4be404b8bca654050fd21805353/src/Mod/Draft/draftguitools/gui_edit.py#L682

Example:
Vector in the `display_tracker_menu` function in gui_edit.py (mouse is over the wire):
```
<pivy.coin.SbVec2s; proxy of <Swig Object of type 'SbVec2s *' at 0x0000009F0C93E370> >
[329, 44]
```
Same vector in the 1st `add_point` function in gui_edit_draft_objects.py (mouse is over the menu):
```
<pivy.coin.SbVec2s; proxy of <Swig Object of type 'SbVec2s *' at 0x0000009F0C93E370> >
[4, 0]
```
In the case of CAD navigation the coords of the vector do not change between `display_tracker_menu` and `add_point`.

The easies solution is to change the referenced code. But there may also be an issue with the navigation modes that needs to be addressed.

`pos = event.getPosition().getValue()`
Or:
`pos = coin.SbVec2s(event.getPosition())`

PR #19925 (2025-03-02): Draft: fix Draft_Edit issue with Gesture navigation
Fixes #19519
commit 6c40857f (2025-03-02): Draft: fix Draft_Edit issue with Gesture navigation

Fixes #19519
closed 2025-03-02 by chennes