FreeCAD/FreeCAD #19526: Creating BIM_project causes  No module named 'mathutils' warning

ErtugrulEge, 2025-02-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings to the Community!

Steps to re-create:
1. Open FC 
2. Create a new BIM file
3. Press BIM_project

```
11:15:06  Note: API not available due to missing dependencies: geometry.add_door_representation - No module named 'mathutils'
11:15:06  Note: API not available due to missing dependencies: geometry.add_railing_representation - No module named 'mathutils'
11:15:06  Note: API not available due to missing dependencies: geometry.add_representation - No module named 'bpy'
11:15:06  Note: API not available due to missing dependencies: geometry.add_window_representation - No module named 'mathutils'
11:15:06  Note: API not available due to missing dependencies: grid.create_axis_curve - No module named 'mathutils'
```

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/81.5973
```

### Subproject(s) affected?

BIM

### Anything else?

It didn't seem like a major problem, but it would be better if the user did not see this type of message.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-02-11:
I can confirm the issue. It does not seem to influence the functionality.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
Installed mods: 
  * symbols_library

```

maxwxyz, 2025-02-11:
@yorikvanhavre FYI, maybe due to the recent change in BIM_project?

yorikvanhavre, 2025-02-12:
The `mathutils` module is a Blender module. This is probably a bug in IfcOpenShell, it should not depend on a Blender module. It does not seem to cause any harm, though, so it's rather cosmetic.

@Andrej730 could you have a look? Would you like us to upstram this bug to IfcOpenShell?

Andrej730, 2025-02-17:
@yorikvanhavre Hi! Those warnings should be removed after https://github.com/IfcOpenShell/IfcOpenShell/commit/774a770c4897d17d3b9b22b60e2944f20c22a702, there's now just `geometry.add_representation` that requires Blender but it's not intended to be used outisde Bonsai, so there's no warning on import for it.

yorikvanhavre, 2025-02-18:
I guess we can magiclose this issue then 😇

closed 2025-02-18 by yorikvanhavre